Arksine/moonraker #215: Question about cors and ports

Kragrathea, 2021-08-23:
I am working on a web app that connects to moonraker via the api and sockets. I just had an issue with cors not working as I expected. It turns out that the port was causing the problem. I am not sure if this is intended behavior or not.

From moonraker.conf
```
cors_domains:
  *.local  #<---Won't match fluiddpi.local:7125/api/version 
  *.local:*   #<---WILL match fluiddpi.local:7125/api/version 
```
This sort of makes sense. But is it the way it is supposed to work?

Arksine, 2021-08-23:
Yes.  In the first entry Moonraker will accept any origin that ends with `.local`.  In the second you included a wildcard for the port, so any origin that ends with `.local` and includes a port will match.

You may have tried `.*local*` and noticed that it will produce an error, because using a wildcard to match a top level domain is unsafe.

Kragrathea, 2021-08-23:
That makes sense. 

I was trying to avoid having to have users modify the moonraker.conf to make my app work. But I think it is going to be required. 

Unless... Could passing the api key could also allow that cors domain somehow? Probably a bad idea but I thought I would ask. :-)

closed 2021-08-23 by Kragrathea