Arksine/moonraker #227: Cura cannot upload with octoprint compatibilty

njoyard, 2021-09-14:
Using Cura 4.10.0.

When sending an upload request, Cura apparently sends an errorneous `path` parameter with value `//`. This results in a HTTP 400 response because in this case moonraker [joins the path with the filename](https://github.com/Arksine/moonraker/blob/8ca953f55c61c6fc67f63133a40f76e97bdbcb89/moonraker/components/file_manager/file_manager.py#L454) (which results in `//somefile.gcode`), which triggers the [root check](https://github.com/Arksine/moonraker/blob/8ca953f55c61c6fc67f63133a40f76e97bdbcb89/moonraker/components/file_manager/file_manager.py#L458).

I know this is just plain bad behaviour from Cura, which octoprint happens to handle graciously (its [path sanitization function](https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/filemanager/storage.py#L1185) will split the path by slashes and join the resulting path elements together, here splitting `//`will result in empty elements, which does not change the path when joining). 

Maybe just adding an explicit check for this case (`path == '//'`) would allow us to upload correctly from Cura while hoping they fix [the bug](https://github.com/Ultimaker/Cura/issues/10453)? If I made such a change, would the PR be accepted?

njoyard, 2021-09-14:
A quick debug lock excerpt from Moonraker to back this:

```
2021-09-14 18:39:22,795 [app.py:post()] -
File Upload Arguments:
path: //
filename: CE3PRO_Reindeer_card_production_version.gcode
tmp_file_path: /tmp/moonraker.upload-100064.mru
Checksum: c8588f04f28d95e2c05656e7e7eb9cf76f047d0bda20cd3d1721636ea70ee616
2021-09-14 18:39:22,798 [web.py:log_exception()] - 400 POST /api/files/local (192.168.1.118): Cannot write to path: //CE3PRO_Reindeer_card_production_version.gcode
2021-09-14 18:39:22,801 [app.py:log_request()] - 400 POST /api/files/local (192.168.1.118) [_API_KEY_USER_] 1143.31ms
~
```

Arksine, 2021-09-14:
We can call `lstrip('/)` on the path.  This is definitely erroneous behavior by cura though, not sure why it would add anything to the path at all.

Arksine, 2021-09-14:
Ok, just pushed the fix.  If you could pull the latest version and give it a test let me know if it fixes the issue.  Thanks.

njoyard, 2021-09-14:
That fixed it. Thanks !

closed 2021-09-14 by njoyard