Arksine/moonraker #228: [power] gpiod import broken on non-debian distros

wlhlm, 2021-09-15:
### Version Information
- klipper https://github.com/KevinOConnor/klipper/commit/8cf1b512233857ec74a4a8fb9ed3245b3daac84b
- moonraker https://github.com/Arksine/moonraker/commit/776e0d6c90461c82bc3153aa81bec976491013b5
- Python 3.9.7
- Archlinux ARM

### Detailed Description
I'm trying to use moonraker's power component, but unfortunately moonraker reports:
> Unable to load gpiod library, GPIO power devices will not be loaded

However, I *do* have gpiod installed:
```
$ python
Python 3.9.7 (default, Sep 10 2021, 12:54:10)
[GCC 10.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gpiod
>>>
```

Digging into the code, I was able to find that moonraker has special handling for importing gpiod:
https://github.com/Arksine/moonraker/blob/776e0d6c90461c82bc3153aa81bec976491013b5/moonraker/components/power.py#L32-L43

It tries to load from `/usr/lib/python3/dist-packages`, but that does not exist on my Archlinux installation:
```
$ ls /usr/lib/python3/dist-packages
ls: cannot access '/usr/lib/python3/dist-packages': No such file or directory
```
There is only a python 3.9-specifc directory. It appears `dist-packages` is a debian-ism.

I can't quite grasp from the [commit](https://github.com/Arksine/moonraker/commit/30155c4a8ccd323b897c0e094d08dd5976982f2b) that introduced this special import handling why it was necessary in the first place. Is this still necessary? If so would it be possible to implement this in a more generic manner that works on Non-Debian distros? I currently need to edit the moonraker source code to be able to use the power component. I found a previous discussion of the `dist-packages` issue, but related to the update manager: #76.

Arksine, 2021-09-15:
Previously a symbolic link to libgpiod was created in the virtualenv.  FWIW, this would have likely failed on arch linux as well without a patch.

I refactored the way the import was handled for a few reasons:
1) Eliminating the symlink simplifies installation
2) There is a "gpiod" package in pip that creates a conflict with the system package.  By prefixing the location of the libgpiod module to `sys.path` Moonraker will always import the correct module
3) In the future it is possible that I will distribute Moonraker using PEX.   Symlinking isn't a option there, so modifying `sys.path` must be used.

I'm ok with adding support for other distributions.  Just let me know where the libgpiod module is located at on arch and I'll loop through various paths until gpiod is found (or isn't).

wlhlm, 2021-09-15:
Thanks for looking into this. I'm installing moonraker through the system package manager a not through a virtualenv, thus moonraker uses sysem-level dependencies located in `/usr/lib/python<python version>/site-packages/`. Here are the python files libgpiod installs:
```
$ pacman -Ql libgpiod | grep python
libgpiod /usr/lib/python3.9/
libgpiod /usr/lib/python3.9/site-packages/
libgpiod /usr/lib/python3.9/site-packages/gpiod.so
```

Arksine, 2021-09-15:
Ok, I have pushed a patch.  It should work for your package installation and through a virtualenv.  Let me know if it solves the issue.  Thanks.

wlhlm, 2021-09-15:
Works like a charm now. Thanks for the quick fix!

closed 2021-09-15 by wlhlm

themacboy, 2021-09-20:
Good morning, 

Not sure if related to this. (im not very skilled on linux systems)

Recently I have installed klipper on a old micro dev computer, Khadas VIM2. Im using Ubuntu 18.04 on it.

Trying to install Moonraker, as explained in Mainsail instructions it always return next error:

"E: Unable to locate package python3-libgpiod"

And personally I can find that library in any repository as described.