Arksine/moonraker #230: UpdateManager - Github rate limit request error

mariocuravic1, 2021-09-21:
moonraker is unable to process github api rate limit request. I've manually updated moonraker (with env rebuild) but the errors is still there. curl to https://api.github.com/rate_limit from the same machine works. Am I missing something?

[moonraker.log](https://github.com/Arksine/moonraker/files/7200763/moonraker.log)

Arksine, 2021-09-21:
You appear to have some kind of networking or system issue.  Its possible that there is some issue with DNS, sometimes requests from within python will timeout even though curl works.  If you haven't attempted to reboot the machine I would recommend doing so.

FWIW, your log is showing quite a few unsafe shutdowns:
```
2021-09-21 07:33:52,587 [database.py:__init__()] - Unsafe Shutdown Count: 52
```
I strongly recommend that you safely shut down your Pi before removing power.  Not doing so can lead to file system corruption and all kinds of system issues.  You may find that you need to reimage your OS.  If that doesn't work it would have to be something to do with your network configuration.

mariocuravic1, 2021-09-21:
Thanks for the update. I'm regularly restarting RPI (sometimes it's shutdown just with the mains power switch, as you pointed out) and the problem has been happening for a few weeks now. I can manually pull the latest version from GitHub.
Is there a way to check get GitHub API endpoint from moonraker's python env?

Arksine, 2021-09-21:
You would have to write some python code to be able to test from within python.

FWIW, one thing I could try would be to use Tornado's curl based client implementation rather than the simple native one.  The downside is that it requires additional dependencies, however if it improves behavior of the client then it may be worth it.

Arksine, 2021-09-21:
Ok, I pushed a couple of commits that switch to the curl backend.  If you need to update manually make sure you run the install script so pycurl gets installed:
```
cd ~/moonraker
git pull
cd scripts
./install-moonraker.sh
```

mariocuravic1, 2021-09-21:
Thanks. I had to wait for current print job to finish.I've updated mooonraker and log file is attached.
It complains that api.github.com can't be resolved but in  ssh I can resolve it:
pi@fluiddpi:~ $ nslookup api.github.com
Server:         8.8.8.8
Address:        8.8.8.8#53

Non-authoritative answer:
Name:   api.github.com
[moonraker.log](https://github.com/Arksine/moonraker/files/7205440/moonraker.log)
Address: 140.82.121.5

mariocuravic1, 2021-10-01:
I'm not sure what was exactly broken but after adding api.github.com entry into /etc/ hosts, update manager could resolve api.github.com hostname again. Disabling ipv6 and adding Google nameservers didn't help.

closed 2021-10-01 by mariocuravic1