Arksine/moonraker #231: issue when inputing new auth IP to use klipper screen

samwells2000, 2021-09-25:
@Arksine 

when I input the new ip 127.0.0.1 into the authorisation list in the .conf the service wont start on boot, and needs to be manually started, but when klipper screen trys to connect its give handshake error 401 unauthorised. 

after fresh boot
pi@fluiddpi:~ $ service moonraker status
● moonraker.service - Starts Moonraker on startup
   Loaded: loaded (/etc/systemd/system/moonraker.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

.conf file
[server]
host: 0.0.0.0
port: 7125
enable_debug_logging: False
config_path: ~/klipper_config
temperature_store_size: 600
gcode_store_size: 1000
log_path: ~/klipper_logs

[authorization]
force_logins: True
cors_domains:
  *.local
  *.lan
  *://app.fluidd.xyz

trusted_clients:
  10.0.0.0/8
  127.0.0.0/8
  169.254.0.0/16
  172.16.0.0/12
  192.168.0.0/16
  127.0.0.1/7125
  FE80::/10
  ::1/128

# enables partial support of Octoprint API
[octoprint_compat]

# enables moonraker to track and store print history.
[history]

# this enables moonraker's update manager
[update_manager]
enable_auto_refresh: True

# this enabled fluidd updates
[update_manager client fluidd]
type: web
repo: cadriel/fluidd
path: ~/fluidd

Arksine, 2021-09-25:
Without a log its difficult to determine what the issue is.  I can tell that your entry of `127.0.0.1` is incorrect.  Remove the `/7125` part to allow localhost access.

Arksine, 2021-11-10:
I'm going to assume that this has been resolved due to inactivity.  If you need further assistance, feel free to join us on the Klipper discord in the Moonraker channel.

closed 2021-11-10 by Arksine