Arksine/moonraker #239: Some MQTT status_objects are continuously publishing updates

pedrolamas, 2021-10-09:
After testing the new `status_objects` MQTT functionality, I came to realize that some of the objects here are being updates on every single check for changes.

These are the ones I noticed while the printer was idle (so `print_stats.state == "standby"`):

- toolhead
  - estimated_print_time
- heater_bed
  - temperature
- extruder
  - temperature

I'm not entirely sure why the `toolhead.estimated_print_time` keeps on increasing while nothing is actively printing, but that is the case.

As for the `heater_bed.temperature` and `extruder.temperature`, that is due to the value having a remarkably high precision (something like 10 decimals, I think) so any small difference will be marked as a change that will then be pushed to MQTT.

I've proposed using a precision setting that would allow one to specify how many decimals to use on floats and avoid such a scenario, but that will not solve the current problem with the `toolhead.estimated_print_time`

Arksine, 2021-10-09:
The `estimated_print_time` field will always change, thus if you subscribe to the `toolhead` you will always receive it.

I'll look into the temperatures.  As mentioned on Discord, it might make more sense to change this behavior in Klipper rather than in Moonraker.

Arksine, 2021-10-11:
Klipper would keep the precision internally, we would only round the data reported by `get_status()`.  We could even take it a step further and only round it in data pushed over the unix socket, but I don't the any `get_status` consumer would need more than 2 decimal places of precision.

Arksine, 2021-11-10:
This was fixed in Klipper, so I'm going to close this issue.

closed 2021-11-10 by Arksine

pedrolamas, 2021-11-10:
I don't think that was enough, as "progress" is also showing the same behavior... 😕

[image]

pedrolamas, 2021-11-10:
I've just added a PR to fix the above case: https://github.com/Klipper3d/klipper/pull/4909