Arksine/moonraker #240: [FR] Configure Host Commands

mikejr83, 2021-10-12:
At the moment the host commands Shutdown and Reboot are hardcoded. Example: https://github.com/Arksine/moonraker/blob/master/moonraker/components/machine.py#L98

Modifying the host commands becomes necessary in situations where the host is using an UPS device to power it. Example: Using a Pi UPS hat.

Suggested moonraker config section:

```ini
[host]
shutdown_cmd: sudo /usr/local/bin/x708softsd.sh
```

FrYakaTKoP, 2021-10-13:
@mikejr83  i agree with arksine that such a behaviour should be handled on systemlevel.

i just tested the systemd script described here https://unix.stackexchange.com/a/284773

It also doesn't execute the script if you reboot your pi. If you need help to set it up please contact me (most preferable on the klipper discord "@FrY Sennberg#4645" in #linux or #moonraker channel)

FrY

Arksine, 2021-10-13:
> It seemed that if Moonraker was going to provide the functionality for host functions that it should also allow some configuration of those functions.

Moonraker offers some limited host functionality that is applicable for default installations.  While I understand the desire for customization it remains my view that running configurable scripts exposes a potentially devastating security hazard.  Moonraker could be used to upload a malicious script or binary to the host machine then reconfigured to run it with root privileges.  Most users don't serve their instance on a public IP, but some do.

On the topic of your particular case, the install script for it appears to be [here](https://github.com/geekworm-com/x708/blob/master/install.sh).  From looking at it, your script simply toggles gpio13.  Moonraker can be configured to toggle the GPIO directly with the [power component](https://moonraker.readthedocs.io/en/latest/configuration/#power).  If necessary I could add a timer to the gpio implementation.

mikejr83, 2021-10-14:
While I disagree and am sad that this functionality isn't going to be available, nor am I willing to maintain a fork, I'll just implement the GPIO solution. I've printed the skirt for my Voron 2.4 that has buttons on it. I'll just manually shut it down by wiring a button to the GPIO.