Arksine/moonraker #241: Fix for "sudo_fix.sh Error: visudo: Command not found" on Armbian?

Swift-Tester, 2021-10-13:
Sadly the `sudo_fix.sh` script is intended for CustomPiOS derived distros (such as Octopi).  It isn't surprising that it does not work on Armbian.  I don't own any devices that run Armbian so I can not provide any official support for them.  If someone is able to come up with a script that gives Moonraker permission to run sudo without a password on Armbian I would be happy to review it and merge if it looks good.

_Originally posted by @Arksine in https://github.com/Arksine/moonraker/issues/92#issuecomment-776615280_

What about calling "visudo" via "sudo visudo"? (Like other utilities in this script).
Hope this won't break CustomPiOS compatibility, and this definitely works on my Armbian.

KwadFan, 2021-12-04:
Hi, I think your patch doesnt affect custompios. So, LGTM.

sjqlwy, 2021-12-17:
TRY THIS:
`echo 'export PATH="/sbin:/usr/sbin:$PATH"' >> ~/.bashrc && source ~/.bashrc`

Arksine, 2021-12-22:
I believe that this issue is resolved as the suggestion was merged in #298.

closed 2021-12-22 by Arksine