Arksine/moonraker #257: Seperate update manager aliases from git_repo_name

FrYakaTKoP, 2021-10-31:
We encountered a problem with moonraker-timelapse where the commit link within mainsail was pointing to a wrong url. The issue was that we defined the alias different then the real repository name.

```
[update_manager timelapse]
vs.
[update_manager moonraker-timelapse]
```

For timelapse this wouldn't be a issue, we could just change the docs and we were golden, but in my opinion it would be great if this would not be bound to a specific alias matching the repository name. There is already a example where the alias needs to be different then the real repository name (because the alias is bound to the service name):

[update_manager webcamd]
type: git_repo
path: /home/pi/crowsnest
origin: https://github.com/mainsail-crew/crowsnest.git
primary_branch: nightly

I have a PR ready which would atleast solve the problem with non working links in mainsail, but maybe further separation of the alias and the repository name and service name would be a good idea.

Arksine, 2021-11-01:
I'm not sure why this would be necessary.  The alias is used for two things: Logging/Messaging and to control the systemd unit if necessary.  It won't prevent git repo updates.  I'm not sure what "crowsnest" is, however it should probably install itself as a service named "crowsnest".  It looks to me like it is attempting to work around moonraker's service control filter by naming itself webcamd.

Arksine, 2021-11-01:
Ok, I just looked at PR #258 and it looks like you just want to report the repo name.  I'm fine with that, I was under the impression that you wanted to configure it.

FrYakaTKoP, 2021-11-01:
yeah maybe i would had to word the issue better sorry for generating confusion.

I agree with changing the alias of crowsnest but as alex said it is bound to the service. That's the reason i didn't just made a PR to add the repo name but raised a issue here. As said in my opinion it would be good to decouple alias, repo name and service name, but i also understand that you like to restrict access to services in general. That's why this issue exist so we can discuss this here. It's somewhere between a issue (which is fixed with the PR #258) and a Feature Request for better separation between alias, repo name and service.

Arksine, 2021-11-10:
PR 258 was merged, so I am going to close this.

closed 2021-11-10 by Arksine