Arksine/moonraker #259: [FR] Endpoint for zipped config files

cbc02009, 2021-11-01:
Multiple config files is becoming more and more common, and it would help troubleshooting immensely if there was a way to access a users entire config via moonraker.

This is something I'm perfectly happy to try working on (though I've only ever used python for scripting), but I wasn't sure where this new functionality should go. I was thinking in the file_manager component, but just wanted to verify.

theoretically it should be very easy to implement, right? Just register a new endpoint `server/files/zip_config` that points to a _handle_zip_config method of the FileManager class.

FrYakaTKoP, 2021-11-01:
@cbc02009 for troubleshooting klippy.log is best way to see parsed multi file configs, because sometimes it's hard to read through mutliple files and analyze which overwrites which. In the log you see what klipper actually loads, this method have proven to be very successful. 

But i would also like to see a zipped config folder for backup reasons. Ideally with a database dump (exluding security related stuff) so a users have a easy way to backup and restore there setup to a new image, if corrupted.

cbc02009, 2021-11-01:
@FrYakaTKoP there are many issues that can't be diagnosed just with klippy.log, as you only see what gets parsed, and not the source files. Often you do need to look at the printer.cfg file to find some weird syntax issue, or doubles of the same config entry, forgotten include statements, etc.

FrYakaTKoP, 2021-11-01:
agree, but IMO still prefer the log. 

Example: 
Real issue: User has multiple files and have different display declaration in multiple files.
User complains: Display led light up green, but he defined red.
Log shows: parsed config shows defined green.
My conclusion: Klipper does everything right, ask the user if he has multiple files, if so i advice the user to look through all his files to find if he has multiple display declarations. Saves me the work looking through multiple files and maybe the User learns something about his configs (this sadly mostly happens with user blindly coping configs together or following a guide without understanding what they do). My approach in supporting is always teach them to help them self's. If you go through the files, find the mistake and send them the corrected files, they will not learn anything.

But again, also like to see such feature but with another reason (backup&restore)

Arksine, 2021-11-01:
This is possible, however I don't think it can be done exactly how you propose.  We can't return the binary zip file through an endpoint, we would have to zip the files, then serve the result as static file.  I think that the tricky part would be determining which config files to include.  I don't think it would be a good idea to zip the entire config directory because truthfully anything can be in there.

As an alternative, we could create a `/server/files/zip` endpoint that takes a list of source files and a destination.  The source files could be a combined list from any root path accessible to moonraker, so it could include logs and specific config files.  The destination must be in a full access path.  Then the user just uses the file browser to download the zipped file.

Some of this could be automated by clients as well.  For example, it could provide a  dialog where users choose which files to include in a "debug download", then this whole process could be done with one click.

marc-gist, 2022-03-06:
I would love this feature that simply triggers a `tar pcfz backup-<datestamp>.tgz ~/klipper_config/` and then returns the compressed file (which i think you could just have tar send as a buffer rather than an intermediate file.  Although keeping a copy of the file on the system wouldn't be a bad idea.

netweaver1970, 2022-03-06:
+1. An easy way to get a full config backup/download would help a lot. Now I have to keep reminding myself to make manual copies via a SFT program to my local disc, while a single button download would make it easier for people to make a local backup. Not everyone can set up a github page+link for backing up...

ChaosBlades, 2022-10-11:
Yes, this would be better utilized for backup of entire config file directory. Currently I have to individually download 12 different config files every time I want to do a backup. Of course this does not even include the backups already in this folder I would also like to have just in case.

As for using this for troubleshooting I think those rare situations are better left to piecemealing since they are so rare.

It would be a nice stretch goal for this feature request to be able to configure a network share in the moonraker config to automatically zip and upload the config directory to said network share on a weekly/monthly basis. 

As I slowly add to and fine tune my printers and the dozens of hours I have put into doing this. There is an increasing amount of anxiety around a failing SD Card and having the piece of mind that my config is safely on a separate backup server would be very much appreciated.