Arksine/moonraker #261: Silent failure when missing perissions for Klipper API socket has

dudeofawesome, 2021-11-03:
Moonraker will silently fail to connect to Klipper if the socket has incompatible permissions (for example if it is owned by root with 755 and Moonraker is running as an unprivileged user)

It would be nice if Moonraker would write something to the log about not having access to the socket file.

Arksine, 2021-11-04:
Moonraker constantly polls the socket and ignores errors as there could be any number of errors introduced that are resolved when Klipper starts.  Its possible to check for existence of the socket then check for permissions, however in a typical scenario permissions should not be a problem.  Klipper creates the unix socket and is run by the same user as Moonraker.

My assumption is that this is related to your comment #253, where you are trying to expose a unix socket created in another container (and hence part of the reason why it might not be a great idea).

Arksine, 2021-11-10:
This should be resolved in commit 61fab42281da627bc14af2e50ed2e026272e5385, so I'm going to close this issue.

closed 2021-11-10 by Arksine

dudeofawesome, 2021-11-10:
Awesome! Thanks for looking at that.