Arksine/moonraker #262: [FR] Add MQTT Last Will And Testament (LWT) support

pedrolamas, 2021-11-05:
A good practice on MQTT is to have the client set a Last Will And Testament (LWT) that the broker will publish automatically if the device connection terminates in a non-conformant way.

It's like telling the MQTT broker "if I get disconnected for some reason publish this message to this topic".

Paho-MQTT does have support for this as shown [here](https://mntolia.com/mqtt-last-will-testament-explained-with-examples/) so hopefully it should be an easy improvement to add!

Arksine, 2021-11-05:
This seems useful and as you said it isn't difficult to add.  I suppose it comes down to hashing out implementation details, specifically which topic the will should be set to.  Presumably it would make sense to apply this to the status topic given that it is the one clients will be listening to, however we could make it configurable as well.

pedrolamas, 2021-11-17:
This is now complete, so closing this FR now.

closed 2021-11-17 by pedrolamas