Arksine/moonraker #266: GPIO not working correctly in Debian Bullseye / Python 3.9.2

pedrolamas, 2021-11-11:
Until today, I had my RaspberryPi 3B+ with FluiddPi and in, I have a relay on GPIO23 that controls some LED lights.

This is what I have in moonraker.conf:

```
[power Lights]
type: gpio
pin: gpiochip0/gpio23
initial_state: off
```

When the pi starts (or when moonraker restarts) the lights are off until the moment I choose to turn them on via Fluidd interface.

A few hours ago I decided to flash my RaspberryPi with the latest RaspiOs and then used Kiauh to install Klipper and Moonraker.

After this, I stopped both services, restored the configs and Moonraker database, and restarted both services.

When I did this, the lights turned on and a split second afterwards off again!

I noticed that if I restart Moonraker, that is happening every single time!!

I ran the following commands on a python REPL and it works perfectly:
```python
import gpiod
from gpiozero import LED
led = LED(23)
led.on()
led.off()
```

So my guess is that there is some issue with Python on the newest RaspiOS which is Debian Bullseye based.

Arksine, 2021-11-11:
To clarify, other than "on and off" behavior on startup, does the gpio toggle correctly?  If so it sounds like a change in the new gpiod library where the line is initializing high instead of low.  Otherwise there may be a bigger issue at play.

If it works outside of the initial behavior I may be able to find a fix for the issue.  I'll try to flash one of my Pi's with the release to see if I can reproduce.

If the GPIO simply refuses to toggle then `moonraker.log` may be useful.

pedrolamas, 2021-11-11:
> To clarify, other than "on and off" behavior on startup, does the gpio toggle correctly?

Correct! After then initial "off->on->off" at Moonraker startup, I can toggle it without any issue at all as I always did!

I've checked the log and couldn't find any issue at all, but here it is in any case!

[moonraker.log](https://github.com/Arksine/moonraker/files/7523812/moonraker.log)

pedrolamas, 2021-11-11:
Good news: I think I've found the problem!

This causes the issue:

```
import gpiod
chip=gpiod.Chip('gpiochip0')
lines = chip.get_lines([ 23 ])
lines.request(consumer='moonraker', type=gpiod.LINE_REQ_DIR_OUT)
```

Once that last line executes, the light turns on!

However, changing the last line to this, it seems to fix the problem!

```
lines.request(consumer='moonraker', type=gpiod.LINE_REQ_DIR_OUT, default_vals=[ 0 ])
```

PR #267 (2021-11-12): power: improves GPIO initial_state handling
Fixes #266

I can confirm this is working fine on my RaspberryPi 3B+ running latest RaspiOS (Debian Bullseye), but better to ensure it also works on other systems (FluiddPi, Mainsail OS, ...)
closed 2021-11-12 by Arksine