Arksine/moonraker #269: API /machine/device_power/devices return value issue

ikirt, 2021-11-13:
I noticed while trying to parse the /machine/device_power/devices API endpoint in python I kept getting the following error:

```
>>> json.loads(req.text)['result']['devices']['status']
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: list indices must be integers or slices, not str
```

If I take a step back and look at everything returned I see:

```
>>> json.loads(req.text)
{'result': {'devices': [{'device': 'wifi_switch', 'status': 'on', 'locked_while_printing': False, 'type': 'tplink_smartplug'}]}}
```

I noticed the 2 square brackets surrounding the device section which causes the python json module in python to fault.

Reviewing another endpoint which I can successfully parse, /printer/objects/query?print_stats I see the following results:

{
    "result": {
        "status": {
            "print_stats": {
                "print_duration": 0.0,
                "total_duration": 0.0,
                "filament_used": 0.0,
                "filename": "",
                "state": "standby",
                "message": ""
            }
        },
        "eventtime": 5506.406120532
    }
}

Arksine, 2021-11-13:
That is because the `devices` field contains a list of devices.  The appropriate way to handle this in a python client would be something like
```python
devices = json.loads(req.text)['result']['devices']
for dev in devices:
  print(dev['status'])
```
or if you wanted to do a one liner that just returns the status of the first device:
```python
json.loads(req.text)['result']['devices'][0]['status']
```

One thing to keep in mind is different endpoints will have return values that are unique to the state they return.  The API documentation will typically give an example and explanation of what is being returned.

ikirt, 2021-11-13:
Thank you for pointing this out, I will review the API documents again !

Arksine, 2021-12-22:
As there has been no further discussion I'm going to go ahead and close this issue.  Thanks.

closed 2021-12-22 by Arksine