Arksine/moonraker #270: SuperSlicer with supports - layer height mismatch

vajonam, 2021-11-13:
I am seeing an issue with the current version of Super Slicer which generates supports with variable layer heights. 

https://github.com/prusa3d/PrusaSlicer/issues/553

[image]
[Amazing Kup (8)_PETG_0.3mm.zip](https://github.com/Arksine/moonraker/files/7531472/Amazing.Kup.8._PETG_0.3mm.zip)

vajonam, 2021-11-13:
Maybe just 

`count (matches)` here as well? 

https://github.com/Arksine/moonraker/blob/ca27c2cf100c47fcbc52d60aa6d61121775a2d9d/moonraker/components/file_manager/metadata.py#L221

Arksine, 2021-11-13:
Moonraker does not actually report the layer count, Fluidd is making that calculation.  The method you are referring to reports object height, so it is behaving as expected.

We can't deduce the layer count by counting matches as Moonraker doesn't analyze the entire gcode file.  IMO, the proper way handle this would be for the slicer itself to report the layer count in the metadata.  Moonraker could then extract it and report it.

vajonam, 2021-11-15:
I can add a post GCode to add a total layers when the g-code is produced. Any specific format?

vajonam, 2021-11-15:
Will open a issue for SuperSlicer and link it with this.

Arksine, 2021-11-15:
If SuperSlicer implements this then I'm ok with adding support to Moonraker.  I don't need a particular format, but it should probably match the way SuperSlicer reports metadata for other items.

vajonam, 2021-11-15:
I have opened a FR will see how this goes. I will leave this open until we get a resolution on the Super Slicer ticket. https://github.com/supermerill/SuperSlicer/issues/1886

if we assume it will something like, which matches the super slicer format.

```
; total_layer_count = {total_layer_count} 

```

then I can add it the custom G-code section now and we can test this.

vajonam, 2021-11-17:
@Arksine this is now fixed / added to SuperSlicer. Can you also look to add the parsing for this. https://github.com/supermerill/SuperSlicer/issues/1886

Arksine, 2021-11-18:
Ok, I'll look at adding it.   From the commit it appears that the layer count is not going to be presented as you proposed.  It appears that is going be placed at the end of the gcode as:
```
; total layers count = {layer_count}
```
Can you confirm?

vajonam, 2021-11-18:
That is what the commit says. SuperSlicer puts all meta info at the end of the file. So I think that is accurate. However since this will be quite / quick fast we can wait for the release to see how it looks in reality before getting this closed out. 

What are your thoughts on klipper accurately reporting current layer? right now that is guesstimate assuming equal layer heights. is possible for klipper to report the exact layer based on gcode comments? 

I think as the UI get features like g-code preview this will become important to show an accurate representation of the status of the print. 

I will also ask the klipper question in the kipper discord.

vajonam, 2021-11-19:
I can confirm.

`; total layers count = 300` 

to the latest release of SuperSlicer

Arksine, 2021-12-05:
Support for this should be merged.

closed 2021-12-14 by vajonam