Arksine/moonraker #272: Error when loading or refreshing Fluiddpi

bill-orange, 2021-11-15:
Starting a few days ago, I am getting this error message every time I refresh the screen or start Fluiddpi. 

[image]

 I looked in Klipper's log and did not find a problem.  Moonraker's log shows the following error:

> File "/home/pi/moonraker/moonraker/components/authorization.py", line 284, in _handle_list_request
>     if user['username'] == API_USER:
> KeyError: 'username'
> 2021-11-14 10:12:25,451 [app.py:log_request()] - 500 GET /access/users/list (192.168.0.7) [_TRUSTED_USER_] 7.65ms

I assume this is the cause of the 'Internal error message' dialog box.
How can I fix this?  Reinstalling Moonraker and Fluiddpi did not help.

Arksine, 2021-11-15:
Can you provide the full moonraker.log?  Thanks.

Arksine, 2021-11-15:
FWIW, I pushed a commit that will validate user entries in the database.  What I suspect happened is that somehow an invalid user entry made its way into your database.  You log would likely provide more info on why this occurred.

bill-orange, 2021-11-15:
@Arksine Sure.  Here's the log.  Hopefully,  you can see something in the log that I missed.

  
[moonraker.log](https://github.com/Arksine/moonraker/files/7539397/moonraker.log)

Arksine, 2021-11-17:
It appears that an invalid user entry made its way into the database.  Either something made a change to the database that shouldn't have or got corrupted (perhaps by the Pi losing power without a proper shutdown).

The commit that I pushed checks the database on startup and removes invalid entries.  I'm unsure as to what you mean by the underlying authentication problem.  If your user entry got corrupted most likely you need to recreate the user.

bill-orange, 2021-11-17:
Where is the user database located?  I can have a look at it.  

Fluiddpi does not display any users. I added and deleted a user, in a fruitless attempt to get my printer phone app to work.  That is likely the source of the corruption and suggests a Fluiddpu problem.  The phone app has not worked since I switched to Klipper.

In any event, I no longer get an error message and everything works.

closed 2021-11-17 by bill-orange