Arksine/moonraker #273: Power status is incorrectly reported for home assistant.

vajonam, 2021-11-15:
Given the printer is on currently on when I query the status it shows it as off.

When the device is printing. 

`http://{MOONRAKER_API}/machine/device_power/devices`

```
{"result": {"devices": [{"device": "Ender3", "status": "off", "locked_while_printing": false, "type": "homeassistant"}]}}
```

I can see the `/states/` call is being made in moonraker. I tired getting to the bottom of this is on power.py. If you don't have an environment I can test on mine for your. I am also happy to jump on discord to help fix. 

This manifests its way like this
https://github.com/fluidd-core/fluidd/issues/398

vajonam, 2021-11-15:
This Helps

```

diff --git a/moonraker/components/power.py b/moonraker/components/power.py
index 9fe3d10..c076202 100644
--- a/moonraker/components/power.py
+++ b/moonraker/components/power.py
@@ -252,8 +252,9 @@ class PrinterPower:
             status = "on" if state else "off"
         elif isinstance(state, str):
             status = state.lower()
-            if status in ["true", "false"]:
+            if status in ["true", "false","on","off"]:
                 status = "on" if status == "true" else "off"
+                status = "on" if status == "on" else "off"
         if status not in ["on", "off"]:
             logging.info(f"Invalid state received: {state}")
             return

```

vajonam, 2021-11-15:
This fixes the issue.

diff --git a/moonraker/components/power.py b/moonraker/components/power.py
index 9fe3d10..21fc017 100644
--- a/moonraker/components/power.py
+++ b/moonraker/components/power.py
@@ -905,6 +905,7 @@ class HomeAssistant(HTTPDevice):
 
     async def _send_power_request(self, state: str) -> str:
         await self._send_homeassistant_command(state)
+        await asyncio.sleep(1.)
         res = await self._send_status_request()
         return res

If seems that if you query the state right away it returns the old state. sleeping for one second fixes the issue.

Arksine, 2021-11-15:
The issue here is almost certainly with Home Assistant.  What it should do is block the API request until it confirms the device state change, then return the new state in the response.  Their API documentation claims it will block for up to 10 seconds then return the correct state.  Obviously that isn't happening, so the workaround when this was contributed was to request the state after requesting that the device be toggled.

We can sleep as you suggest to work around it, but the delay is likely going to depend on the switch.  What if something goes wrong during the request?  We have no way of knowing if we need more delay or if Home Assistant simply can't toggle the switch.  This is not an ideal solution.

The correct thing to do is probably to disable support for HA switches in Moonraker until Home Assistant fixes this on their end.  I know that this would disappoint some people, so I'm hesitant to do it.  Instead I might introduce a configurable delay and add a warning to Moonraker that the Home Assistant side is broken.

vajonam, 2021-11-15:
Agree with you 100%. 

Another integration we can add is mqtt, which will also capture out of band changes to state since we can listen on that channel. I know we have outbound support for mqtt, does the infa allow for inbound listening for messages, if you give me a pointer as to what I can see for inbound mqtt.

I can contribute a PR with mqtt support for power.

vajonam, 2021-11-15:
UPDATED diff with configurable "status_delay"

vajonam, 2021-11-15:
Thanks fixed with latest!

closed 2021-11-15 by vajonam