Arksine/moonraker #275: publish M118 to MQTT

FotoFieber, 2021-11-15:
Would be nice to have a possibility to publish M118 messages to MQTT under e.g.
**my_printer/moonraker/serial/**
to integrate klipper into my home automation setup. First use case would be to mimic action commands from octoprint.

My current solution
https://github.com/FotoFieber/KlipperSerialMQTT
is not really nice but working. :)

P.S. Thx all for the awesome work!

pedrolamas, 2021-11-26:
I know it's not what you asked for, but you can get at least the `M117` output via MQTT with this:

```
[mqtt]
address: mqtt_server
username: mqtt_user
password_file: ~/.mqtt_pass
instance_name: printer_name
status_objects:
  display_status=message
```

FotoFieber, 2021-11-27:
@pedrolamas Thx, I used your proposed solution but found it too ugly. Cryptic commands in the display are not nice.

Arksine, 2021-12-23:
I thought about this, and while it is possible to publish gcode responses I'm not sure that is the best way to accomplish your goal.  Generally speaking gcode responses aren't very useful as most of the data they contain can be queried with structured results.  The exception are the live results displayed during calibration procedures, however I'm not sure that is useful for IoT.

It occurred to me that perhaps the better way to accomplish your goal is to add the ability to publish topics from klipper.  For example, in your case we could add the following to printer.cfg:
```ini
[gcode_macro M118]
rename_existing: M118.1
gcode:
  {% set payload = {"message": rawparams} %}
  {action_call_remote_method("publish_mqtt_topic",
                             topic="klipper/echo",
                             payload=payload,
                             use_prefix=True)}
  M118.1 {rawparams}
```
If your instance name is `my_printer`, the payload would be published to the `my_printer/klipper/echo` topic.  You could send `rawparams` directly rather than include it in an object.  [Click here](https://moonraker.readthedocs.io/en/latest/configuration/#publishing-topics-from-klipper) for the full documentation.

FotoFieber, 2021-12-24:
Thx very much. Your solution is perfect, working really fine. I appreciate the time you invested to answer this issue.

closed 2021-12-24 by FotoFieber