Arksine/moonraker #278: Issue with Prusa Slicer and Moonraker

chickenmcniggels, 2021-11-17:
Whenever I try to upload to Moonraker this error shows up.. The slicer says that it can communicate with OctoPrint, when I hit that test button but whenever I hit upload this error pops up

[image]

I've searched the entire Internet and cannot find an answer to this problem. Is someone familiar with this error?

Afterschnitzel, 2021-11-22:
check your printer.cfg if you have the right path to gcode folder:

`[virtual_sdcard]
path: ~/gcode_files`

Afterschnitzel, 2021-11-22:
[image]

chickenmcniggels, 2021-11-23:
This fixed it! Thank you

Arksine, 2021-11-26:
Since this appears to be fixed I am closing this issue.

closed 2021-11-26 by Arksine