Arksine/moonraker #289: [FR] Add secrets file

pedrolamas, 2021-11-26:
Some of us have passwords and host names on the moonraker config file that ideally should not be exposed either on the logs (that we share for support/debug purposes) or even when we share our configs in GitHub or elsewhere.

What I propose is something like Home Assistant and ESPHome have for secrets: https://www.home-assistant.io/docs/configuration/secrets/

Arksine, 2021-12-28:
This is now available, the documentation is [here](https://moonraker.readthedocs.io/en/latest/configuration/#secrets).

closed 2021-12-28 by Arksine

pedrolamas, 2021-12-28:
@Arksine secrets is working perfectly for username and password - thanks for that - but what do you think about adding at least for server names also? I ask this as I use a registered domain name as a local one (so my devices are `device.local.mydomain.com`) and would prefer not to expose the public domain!

Arksine, 2021-12-28:
Understood.  There are some options that take an address where I could add support for secrets, however there are some where I cannot.  For example, it is necessary to see what is in the `cors_domains` option to troubleshoot it.  Also it is likely that in some cases the supplied address gets logged.