Arksine/moonraker #291: Moonraker Down

yasaslive, 2021-11-28:
Moonraker is down and unable to start after latest update!
  
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/moonraker.py", line 238, in load_component
    load_func = getattr(module, "load_component")
AttributeError: module 'components.job_queue' has no attribute 'load_component'
2021-11-28 06:32:53,921 [moonraker.py:main()] - Moonraker Error
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/moonraker.py", line 238, in load_component
    load_func = getattr(module, "load_component")
AttributeError: module 'components.job_queue' has no attribute 'load_component'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/moonraker.py", line 835, in main
    server = Server(app_args, file_logger, event_loop)
  File "/home/pi/moonraker/moonraker/moonraker.py", line 147, in __init__
    self._load_components(config)
  File "/home/pi/moonraker/moonraker/moonraker.py", line 219, in _load_components
    self.load_component(config, component)
  File "/home/pi/moonraker/moonraker/moonraker.py", line 246, in load_component
    raise ServerError(msg)
utils.ServerError: Unable to load component: (job_queue)
2021-11-28 06:32:53,922 [base_events.py:default_exception_handler()] - Task was destroyed but it is pending!
task: <Task pending coro=<INotifyHandler._notify_root_updated() running at /home/pi/moonraker/moonraker/components/file_manager/file_manager.py:1059>>
2021-11-28 06:32:53,929 [base_events.py:default_exception_handler()] - Task was destroyed but it is pending!
task: <Task pending coro=<INotifyHandler._notify_root_updated() running at /home/pi/moonraker/moonraker/components/file_manager/file_manager.py:1059>>
2021-11-28 06:32:53,930 [moonraker.py:main()] - Server Shutdown

bill-orange, 2021-11-28:
Try it now.  Please see my issuue #290.  Github is back up.

yasaslive, 2021-11-29:
Reversing back to previous commit was my solution

closed 2021-11-29 by yasaslive