Arksine/moonraker #292: Unable to load component: (gpio)

simpat1zq, 2021-11-30:
I get this error in the logs, and it causes Fluidd to show a warning. I'm running Linux Mint an a PC, and I'm not even using GPIOs for anything. Is there anything I can do here?

2021-11-30 06:55:25,920 [moonraker.py:load_component()] - Component (job_queue) loaded
2021-11-30 06:55:25,922 [moonraker.py:load_component()] - Unable to load component: (gpio)
Traceback (most recent call last):
  File "/home/hass/moonraker/moonraker/moonraker.py", line 239, in load_component
    component = load_func(config)
  File "/home/hass/moonraker/moonraker/components/gpio.py", line 135, in load_component
  File "/home/hass/moonraker/moonraker/components/gpio.py", line 24, in __init__
    self.gpiod: Any = load_system_module("gpiod")
  File "/home/hass/moonraker/moonraker/utils.py", line 199, in load_system_module
    raise ServerError(f"Unable to import module {name}")
utils.ServerError: Unable to import module gpiod
2021-11-30 06:55:25,922 [power.py:__init__()] - Power component loading devices: ['power printer']
2021-11-30 06:55:25,922 [app.py:register_local_handler()] - Registering HTTP Endpoint: (GET) /machine/device_power/devices

pedrolamas, 2021-12-01:
Considering that I can see this:

> File "/home/hass/moonraker/moonraker/components/gpio.py", line 135, in load_component

That tells me you are using an out of date moonraker, as currently [gpio.py](https://github.com/Arksine/moonraker/blob/master/moonraker/components/gpio.py) has 130 lines!

Please update moonraker to the latest version and see if that fixes the problem!

simpat1zq, 2021-12-02:
[image]
According to the screenshot it's up to date. Can you confirm that is the latest version?

Arksine, 2021-12-02:
From what I gather you are using a desktop PC with no GPIOs available.  However Moonraker is attempting to load a gpio power device.  Check `moonraker.conf` and remove the `[power printer]` section, as that appears to be the culprit.

simpat1zq, 2021-12-02:
Ok, so removing that gets rid of the warning. But I am using the TPLinj module, not GPIO. Is it incorrectly trying to load the GPIO module even though it isn't needed?

Arksine, 2021-12-02:
It shouldn't load the gpio module if you have it configured for a tplink device.  Upload `moonraker.log` and I'll take a look at it.

simpat1zq, 2021-12-02:
Thanks. Here you go.
[moonraker.log (1).txt](https://github.com/Arksine/moonraker/files/7637937/moonraker.log.1.txt)

Arksine, 2021-12-02:
Ok.  It appears that somehow the latest version of moonraker got pulled without the service restarting.  Unfortunately I don't think fluidd actually restarts the service via the UI (it restarts the server which is different, it doesn't kill the program an reload it).

SSH into your pi an run:
```
sudo service moonraker restart
```
This should allow the tplink plugin to load without the warning.

simpat1zq, 2021-12-02:
Ah. That was it. So updating through fluidd requires a manual service restart it seems.  That is good to know. Thank you for your help.

closed 2021-12-05 by Arksine