Arksine/moonraker #293: Moonraker disconnects

webel69, 2021-12-01:
Hi, 
I am having a problem with random moonraker disconnects both on a Raspberry pi 4 running fluidpi and also a raspberry 2 running the same.  On the pi4 the issue started after updating from Fluidpi 16.2 using the Fluiddpi dashboard. After the update the disconnects were random., but made uploading gcode from the web ui extremely difficult.   
I started from scratch installing 16.2 on the raspberry pi2 with a fresh install and even reflashed klipper on the ender 3 max. Crashes wew much less frequent, but It still has issues when uploading gcode. with in a few seconds of the window opening up to select the gcode file to upload moonraker disconnects..  I did update moonraked acording to another post relating to diconnects.  I have attached the most recent log.  Hope you can help.
[moonraker.log](https://github.com/Arksine/moonraker/files/7630804/moonraker.log)

Arksine, 2021-12-02:
This log is an older version of moonraker, but it appears you are having local networking issues.  I don't see where fluidd tries to connect, but what I do see is a failure for to reach `api.github.com`.  Make sure your wifi signal is good, if possible try connecting via ethernet to see if it resolves the issue.

webel69, 2021-12-02:
There were multiple moonraker logs,  not in front of the computer now, which do you need?

Arksine, 2021-12-02:
The other logs are likely just rollovers. If you uploaded the latest log that is ok.  According to the log, the verision of Moonraker on your system is nearly 200 commits behind master.  Also, its having a problem reaching github, which is either a wan issue or a DNS issue.

Arksine, 2021-12-20:
As there has been no further comment I'm going to close this issue.  If you need further assistance please join the Klipper discord.    You can request help in the `moonraker` channel.  Though as mentioned above, this appears to be a local networking issue.

closed 2021-12-20 by Arksine