Arksine/moonraker #302: update_manager

boubouexpress, 2021-12-07:
### What happened

Hello
since an update, I can no longer activate in moonraker.conf [update_manager] and [update_manager client fluidd]
if I activate them I can no longer connect

### What did you expect to happen

that I connect to the printer via explorer with [update_manager] activate

### How to reproduce

just activate the lines in moonraker.conf

[update_manager]
enable_auto_refresh: True

[update_manager client fluidd]
type: web
repo: cadriel/fluidd
path: ~/fluidd

### Additional information

[update_manager]
enable_auto_refresh: True

[update_manager client fluidd]
type: web
repo: cadriel/fluidd
path: ~/fluidd

Arksine, 2021-12-08:
Please attach your `moonraker.log`. On the surface this sounds like a networking issue.

boubouexpress, 2021-12-09:
after a last update of the pi system is a restart of the pi it is back to normal

merci

closed 2021-12-09 by boubouexpress