Arksine/moonraker #303: Update Frequency Adjustment

TehProgrammer, 2021-12-08:
### Is your feature request related to a problem? Please describe

Not currently an issue, but a quality of life thing.  Klipperscreen and Fluidd both are a bit awkward with a 1hz update frequency.

### Describe the solution you'd like

I want a configuration option to change from the default update frequency.  There would obviously be a valid range.  1hz (default) to 10hz (max).  I don't see a reason to allow more than 10hz.  30hz would probably be to absolute max before it gets ridiculous.

### Describe alternatives you've considered

There does not seem to be any alternatives short of forking and changing the code myself.  I'm very much too lazy to dedicated myself to this as I'd have to actively keep up with new releases too.

### Additional information

This could be added under and [advanced] config category, with the tag of update_rate:

Just a suggestion.

Also, if this is planned / already a thing, please point my to the documentation and close this.  I did not find it despite my searching.  Thanks!

Arksine, 2021-12-08:
The clients decide their update frequency.  Updates to data wouldn't be available at 10Hz though, for example Klipper pushes data every 250ms, Moonraker forwards that over the websocket.

Data that Moonraker collects gets pushed over the websocket when it changes.

TehProgrammer, 2021-12-10:
Is there a way to increase Klipper's frequency then?  Just curious.  I really like smooth / not choppy readings.

Arksine, 2021-12-12:
It isn't configurable.  However as mentioned you would need to modify KlipperScreen/Fluidd to achieve faster than 1Hz refresh rates.  I would start there, see how 250 ms works, then if you want you could fork Klipper and increase the rate at which updates are pushed.

Arksine, 2021-12-20:
As there is no further comment I am going to close this issue.  It isn't possible to increase the update frequency, however it is possible for the client to sample faster than 1Hz...however that would require modifications to the client.

closed 2021-12-20 by Arksine