Arksine/moonraker #304: Paneldue heaters are in wrong spots on paneldue when using multiple extruders

jan3ll3, 2021-12-08:
### What happened

I am converting a toolchanger that uses the e3d system to klipper. Ive worked out just about all the scripts to have it behave like my other toolchangers running RRF. I wanted the displays to match so the users at my shop can just look at the screen and not know its any different. 

When I set a temperature on the display it goes to the wrong tool. Everything has shifted one register to the right. The bed temperature is under extruder (t0), extruder is on extruder1 (t1) and extruder1 is on the bed. 

Would love any documentation to help set up the paneldue correctly. Cant seem to find anything to configure. 

Additionally when i press the button to load t1 (extruder1) it sends the command T-1 which I cannot make a klipper macro for due to the minus symbol

### What did you expect to happen

Temperatures should be in correct location
t1 button should emit T1 command 

### How to reproduce

Followed instructions to load paneldue and everything seemed fine until i added the second tool to the system

Arksine, 2021-12-08:
Ok.  I suspect that the heater problem is a sorting issue, I think I can resolve that quickly.  I'll need to dig into the "T-1" issue, it seems strange that the PanelDue would insert a dash into the command.  I'll see if I can set up a fake secondary extruder to reproduce.

jan3ll3, 2021-12-08:
Thank you for the quick reply, and for looking into this!

Figured I should probably add my logs:
[moonraker.log](https://github.com/Arksine/moonraker/files/7681025/moonraker.log)
[klippy.log](https://github.com/Arksine/moonraker/files/7681026/klippy.log)

jan3ll3, 2021-12-08:
Also T-1 is a command of RRF for unload all tools, really any T# that is not a designated tool does this, so some people will say T99 but it is commonly T-1

Arksine, 2021-12-09:
I pushed some changes in commit 54abd2748652b431469f18a46698f1223fffd4e1.  I believe that this will fix the sorting issue and properly assign the correct tools. 

I had to dig into the PanelDue code to get an idea as to why its sending `T-1`, and I think I have it figured out.  Essentially in a multi-extruder setup, only the selected extruder can have an "active" status, the others have to be set as "standby".  Attempting to change to a tool set as "active" sends the T-1 gcode.

I made this change, however I can't test it.  If you could give it a try and tell me if its working correctly that would be great.

jan3ll3, 2021-12-10:
Both problems fixed! Thank you so much for the quick response! 

Is there anything I can read to setup dialog options on the paneldue? On RRF I have macros for setting Z probe calibration, but I cant seem to figure out how to start a dialog box with klipper

Arksine, 2021-12-12:
It isn't currently possible to create a custom dialogs through Moonraker.  You can create Macros though.

It should only send `T-1` if you select the active extruder, at least according to the PanelDue source code.  I'll try to take another look at it, unfortunately I haven't found a simple way to fake a second extruder to test this myself.

jan3ll3, 2021-12-13:
Thank you again for your help! 

I may dig a bit into this on the paneldue firmware, if I do I'll post here

jan3ll3, 2021-12-13:
Yeah I found it in the 1.14 paneldue firmware. File Paneldue.cpp line 728. I'll compile later, but im certain that i can just change it. Its a string it sends, and I can just change the string. 

https://github.com/jan3ll3/PanelDue/tree/1.14Klipper

Arksine, 2021-12-13:
That might work until I can figure out why the extruder is showing up as active when I set the hstat field to standby.  FWIW, Moonraker's PanelDue module is intended to work with PanelDue version 1.24.  I don't know if there is an appreciable difference, but you may want to flash 1.24 to see if it works before trying a modified version.

jan3ll3, 2021-12-13:
oops! I knew it ended in a 4, mental typo. will fix

jan3ll3, 2021-12-13:
Location of edit changed, but its still a string. Changed from "T-1" to "TX"

https://github.com/jan3ll3/PanelDueFirmware/tree/1.24-Klipper

tfjield, 2022-07-10:
@jan3ll3 Did you ever compile the PanelDue edits you made?