Arksine/moonraker #305: Moonraker Klipper connection Issues

HermanGaviriaMarin, 2021-12-09:
### What happened

After successfully imaging a drive with Fluiddpi and entering the web interface of Fluidd/Mainsail, making sure that everything is up to date (updated using the web controls and KIAUH), making sure that the config file was setup with my specific set up, and making sure that the main board was flashed with appropriate firmware, I simply can't get a connection with klippy.  

[image]

[moonraker.log](https://github.com/Arksine/moonraker/files/7681334/moonraker.log)

[image]

 for some reason my Klippy log also stopped updating

### What did you expect to happen

Moonraker should connect with Klipper or at least provide information as to how to resolve issues (which it isually does but this time it didnt't). After successful installation and connection printer should be functional. 

### How to reproduce

1.  Download latest version of FluiddPi, flash it on SD card, change supplicant (do all the normal steps)
2.  Enter web interface (Fluidd) update config file and make sure macros are present within config for Moonraker or use KIAUH macros
3. Update everything that needs to be updated (klipper, OS, moonraker, etc)
4. connect (spoiler, it doesn't, at least for me)
5. ask for help 

### Additional information

Printer: Creality Ender 3 v2
PI Model: 4B 4GB

  Maybe Relevant Modifications: 

- BTT Octopus board V1.1 
- TMC 2209 with sensorless homing on XY
- CR-Touch 
- Filament Sensor

Arksine, 2021-12-09:
This isnt a bug in Moonraker.  It can't connect to Klippy because Klippy is not running.  Something is wrong with your Klipper installation and you will need to determine what it is and get it up an running.  The Discord is a good place to get help with Klipper.

Arksine, 2021-12-20:
As there has been no further comment I'm going to close this.  If you still need help, please join the Klipper discord.  There we can point you to the correct channel.  Thanks.