Arksine/moonraker #306: Update failed

Talha909, 2021-12-09:
### What happened

I try to update the moonraker but it failed. 

I try to update moonraker from
v0.7.1-225-g1ad83ce to -> v0.7.1-228. I got this error.

Updating Application moonraker...
Application moonraker: Updating Repo...
error: object file .git/objects/f5/f87bf08c9d8bf1ca4c2b944dd5ea428724f699 is empty
error: object file .git/objects/f5/f87bf08c9d8bf1ca4c2b944dd5ea428724f699 is empty
fatal: loose object f5f87bf08c9d8bf1ca4c2b944dd5ea428724f699 (stored in .git/objects/f5/f87bf08c9d8bf1ca4c2b944dd5ea428724f699) is corrupt
fatal: the remote end hung up unexpectedly
Error updating moonrake

### What did you expect to happen

Update on new version. 

### How to reproduce

I installed the fluidd and click on update moonraker . but it gave me the error. 
[printer.txt](https://github.com/Arksine/moonraker/files/7685043/printer.txt)
[fluidd-access.log](https://github.com/Arksine/moonraker/files/7685044/fluidd-access.log)
[klippy.log](https://github.com/Arksine/moonraker/files/7685045/klippy.log)
[moonraker.log](https://github.com/Arksine/moonraker/files/7685047/moonraker.log)

### Additional information

I try to update moonraker from
v0.7.1-225-g1ad83ce to -> v0.7.1-228. I got this error.

Updating Application moonraker...
Application moonraker: Updating Repo...
error: object file .git/objects/f5/f87bf08c9d8bf1ca4c2b944dd5ea428724f699 is empty
error: object file .git/objects/f5/f87bf08c9d8bf1ca4c2b944dd5ea428724f699 is empty
fatal: loose object f5f87bf08c9d8bf1ca4c2b944dd5ea428724f699 (stored in .git/objects/f5/f87bf08c9d8bf1ca4c2b944dd5ea428724f699) is corrupt
fatal: the remote end hung up unexpectedly
Error updating moonrake

Arksine, 2021-12-11:
You will need to do a hard recovery.  Its possible the the client you are using exposes this functionality, or you can do it manually via ssh:
```
sudo service moonraker stop
cd ~
rm -rf moonraker
git clone https://github.com/arksine/moonraker.git
./moonraker/scripts/install-moonraker.sh
```
FWIW, the corruption is likely the result of this ```
2021-12-09 14:42:12,370 [database.py:__init__()] - Unsafe Shutdown Count: 126
```
Make sure to safely power down your machine prior to removing power if at all possible.  Not doing so often leads to filesystem and repo corruption.

Arksine, 2021-12-20:
I'm going to close this issue as there has been no further comment.  If you continue to have issues I would suggest joining the Klipper discord, where someone can walk you through the procedure for recovery.  Thanks.

closed 2021-12-20 by Arksine