Arksine/moonraker #309: Death loop since ae4c4ad6

RomRider, 2021-12-11:
### What happened

Since commit ae4c4ad6786bea57b36a8bc5d36a0254bbd03973, when the mcu is disconnected from klipper and then reconnected, there's a death loop happening here:
```
Dec 12 00:23:37 vcore3 python[28846]: ConnectionResetError: [Errno 104] Connection reset by peer
Dec 12 00:23:37 vcore3 python[28846]: [moonraker.py:_read_stream()] - Klippy Stream Read Error
Dec 12 00:23:37 vcore3 python[28846]: Traceback (most recent call last):
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
Dec 12 00:23:37 vcore3 python[28846]:   File "/usr/lib/python3.7/asyncio/streams.py", line 533, in readuntil
Dec 12 00:23:37 vcore3 python[28846]:     raise self._exception
Dec 12 00:23:37 vcore3 python[28846]:   File "/home/pi/moonraker/moonraker/moonraker.py", line 743, in _read_stream
Dec 12 00:23:37 vcore3 python[28846]:     data = await reader.readuntil(b'\x03')
.... infinitely
```

### What did you expect to happen

For moonraker to not loop :)

### How to reproduce

1. Turn of the mcu
2. Turn on the mcu
3. Moonraker start to loop taking 100% CPU and filling in the logs (tenths of MB per seconds)

### Additional information

Rolling back to 69a06dd1 make it work again.

Arksine, 2021-12-12:
I pushed a commit that should resolve this.  There is something very strange going on with your system though, disconnecting the MCU should not disconnect the Unix Socket.  Could you pull the latest version, reproduce (it shouldn't fill up the log this time), then upload `moonraker.log` and `klippy.log`?  I'd like to see how powering down an MCU is causing the Unix socket to reset, I can't reproduce that behavior.

RomRider, 2021-12-12:
It works with the fix, thanks!

I can see that I have this new exception that you have introduced `Error closing Klippy Unix Socket`. Seems like that was the issue before.

I have sanitized the files (tokens and stuff...) and it's clean files:
1. starting klipper service
2. starting moonraker service
3. powering on the mcu
4. powering off the mcu
5. powering on the mcu again

[moonraker.log](https://github.com/Arksine/moonraker/files/7698773/moonraker.log)
[klippy.log](https://github.com/Arksine/moonraker/files/7698774/klippy.log)

Arksine, 2021-12-20:
I'm going to go ahead and close this as resolved in Moonraker.  I will still continue to look into Klippy to see if I can determine what is causing the connection to drop.  Thanks.

closed 2021-12-20 by Arksine

comatose-tortoise, 2021-12-22:
When will this fix be released? I'm having this all the time now and I don't know how to update outside of the GUI (i.e., pull the latest vesion).

Arksine, 2021-12-22:
The fix has been released.  To manually install SSH into your host (I'm assuming its a Rapsberry Pi) and run the following commands:
```
sudo service moonraker stop
cd ~/moonraker
git pull
~/moonraker/scripts/install-moonraker.sh
```
The last command will make sure that all new dependencies are installed.  It should also be possible to update using the UI if you want to give that a try.  Both Mainsail and Fluidd have a "refresh" button that forces Moonraker to check for the latest updates.  Click that and it should report the latest version, allowing you to update your machine.

comatose-tortoise, 2021-12-22:
> The fix has been released.  To manually update SSH into your host (I'm assuming its a Rapsberry Pi) and run the following commands:

Thank you very much! I will try this.

commit 31dd758d (2021-12-12): moonraker: unix socket fix

Handle connection errors in the read loop.  Set a maximum number
of consecutive errors encountered during a read before aborting.
Resolves #309.