Arksine/moonraker #311: Moonraker changes permissions, even if the file is already accessible by it

cab404, 2021-12-13:
### What happened

After edit through moonraker interface, it changed group and user on printer.cfg from "klipper" to "moonraker", making it inaccessible by klipper. "moonraker" is in group "klipper", but ran with group "moonraker".

### What did you expect to happen

No permission changes, or configuration flag for that behavior.

### How to reproduce

Explaind in "What happened"

Arksine, 2021-12-14:
This is because you are using a non-standard installation that runs Moonraker with a "moonraker" user and Klipper with a "klipper" user.  When Moonraker modifies a file what actually happens is the file is uploaded to a temporary directory, which is then moved to the desired location.  There is no way around this, so the behavior you describe expected.  The solution is to make sure that Klipper and Moonraker are run with the same user and group.

cab404, 2021-12-14:
But there is a way around it. Just open that same file and write to it in discard mode.

Arksine, 2021-12-14:
That would be incredibly slow for large files on SBCs.  It would be possible to add a flag to the upload endpoint that does this, however clients would need modification for it to work.  Generally speaking I'm disinclined from adding complexity to support non-standard installations though.

cab404, 2021-12-15:
Well, I'll find another way to separate those users -- probably adding klipper to moonraker group instead. Thank you for answering and making Moonraker!)

closed 2021-12-15 by cab404