Arksine/moonraker #314: Add Domoticz support to power control

TheLion, 2021-12-19:
### Is your feature request related to a problem? Please describe

No.

### Describe the solution you'd like

Add Domoticz support to power control.

I tried to do it myself in power.py, but I'm not that good with Python :-(

### Additional information

Domoticz uses the following command to switch on/off a switch:

On: http://[IP]:[PORT]/json.htm?type=command&param=switchlight&idx=[ID]&switchcmd=On
Off: On: http://[IP]:[PORT]/json.htm?type=command&param=switchlight&idx=[ID]&switchcmd=On

The status is queried by reading the result.Status in JSON on http://[IP]:[PORT]/json.htm?type=devices&rid=[ID]

This looks like:
```
{
	"ActTime" : 1639914744,
	"AstrTwilightEnd" : "18:37",
	"AstrTwilightStart" : "06:41",
	"CivTwilightEnd" : "17:11",
	"CivTwilightStart" : "08:06",
	"DayLength" : "07:42",
	"NautTwilightEnd" : "17:55",
	"NautTwilightStart" : "07:22",
	"ServerTime" : "2021-12-19 12:52:24",
	"SunAtSouth" : "12:39",
	"Sunrise" : "08:48",
	"Sunset" : "16:30",
	"app_version" : "2021.1 (build 13872)",
	"result" : 
	[
		{
			"AddjMulti" : 1.0,
			"AddjMulti2" : 1.0,
			"AddjValue" : 0.0,
			"AddjValue2" : 0.0,
			"BatteryLevel" : 255,
			"CustomImage" : 4,
			"Data" : "On",
			"Description" : "",
			"DimmerType" : "none",
			"Favorite" : 0,
			"HardwareDisabled" : false,
			"HardwareID" : 2,
			"HardwareName" : "Dummy",
			"HardwareType" : "Dummy (Does nothing, use for virtual switches only)",
			"HardwareTypeVal" : 15,
			"HaveDimmer" : true,
			"HaveGroupCmd" : true,
			"HaveTimeout" : false,
			"ID" : "000143A3",
			"Image" : "Printer",
			"IsSubDevice" : false,
			"LastUpdate" : "2021-12-19 12:52:18",
			"Level" : 0,
			"LevelInt" : 0,
			"MaxDimLevel" : 100,
			"Name" : "RatRig V-Core-3",
			"Notifications" : "false",
			"PlanID" : "0",
			"PlanIDs" : 
			[
				0
			],
			"Protected" : false,
			"ShowNotifications" : true,
			"SignalLevel" : "-",
			"Status" : "On",
			"StrParam1" : "aHR0cDovL3NoZWxseXBybzRwbS04NGNjYTg3ZWFmZWMuZm91cnBldHMubmV0L3JlbGF5LzE/dHVybj1vbg==",
			"StrParam2" : "aHR0cDovL3NoZWxseXBybzRwbS04NGNjYTg3ZWFmZWMuZm91cnBldHMubmV0L3JlbGF5LzE/dHVybj1vZmY=",
			"SubType" : "Switch",
			"SwitchType" : "On/Off",
			"SwitchTypeVal" : 0,
			"Timers" : "false",
			"Type" : "Light/Switch",
			"TypeImg" : "lightbulb",
			"Unit" : 1,
			"Used" : 1,
			"UsedByCamera" : false,
			"XOffset" : "0",
			"YOffset" : "0",
			"idx" : "851"
		}
	],
	"status" : "OK",
	"title" : "Devices"
}

```

olivierberthomme, 2023-10-26:
I could implemented Domoticz support for power module through http. See doc #732.

TheLion, 2023-10-26:
That would be great!

olivierberthomme, 2023-10-26:
It's already done ; you have to customize (domoticz url / user base64) as explained in my doc commit

TheLion, 2023-10-28:
I'll take a look and give it a try, thanks!

TheLion, 2024-02-24:
Does work with Domoticz. But switched to Home Assistant, so not using it any more ;-)

closed 2024-02-24 by TheLion