Arksine/moonraker #316: trusted_clients - comments

docgalaxyblock, 2021-12-22:
### What happened

Since some commit this week this is not working anymore
```
[authorization]
trusted_clients:
    X.X.X.X   #moonraker-monitor
```

[image]

it's only a little QOL probem

### What did you expect to happen

that moonraker does not try to read it as part of an hostname

### How to reproduce

comment anything after an IP

### Additional information

Happy 🎄

Arksine, 2021-12-22:
Hi.  Actually your trusted IP wasn't working before, the difference is Moonraker wasn't warning you about it. Instead Moonarker was assuming that it was a domain name .  A recent change now validates that domain names supplied into the `trusted_clients` are properly structured.

Python's configparser [doesn't support inline comments](https://docs.python.org/3/library/configparser.html#supported-ini-file-structure), each comment must be on its own line.

Arksine, 2022-01-06:
As discussion seems to have concluded I am going to close this.

closed 2022-01-06 by Arksine