Arksine/moonraker #317: Component 'power' failed to load with error: Error Refeshing Device Status: printer

grantemsley, 2021-12-23:
### What happened

After restarting moonraker, or on first boot, sometimes get a warning:

> An error was detected while loading the moonraker components. Please check the logfile and fix the issue.
> Following plugin has an error:
> power
> Component 'power' failed to load with error: Error Refeshing Device Status: printer

The Tasmota plug it's configured for seems to be working perfectly and I have no issues accessing it's web interface.

Maybe related, maybe not, but I only started seeing this issue after adding a WLED device as well.

### What did you expect to happen

No error and power control working.

### How to reproduce

Happens intermittently.  Restarting moonraker will sometimes clear the error, or sometimes cause it come back.

### Additional information

[moonraker (3).log](https://github.com/Arksine/moonraker/files/7769541/moonraker.3.log)

The relevant part of the logs seems to be:

```
2021-12-23 12:13:23,937 [wled.py:_send_wled_command()] - WLED: Error HTTP 599: Unknown
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/wled.py", line 144, in _send_wled_command
    response = await self.client.fetch(request)
tornado.curl_httpclient.CurlError: HTTP 599: Unknown
2021-12-23 12:13:23,937 [wled.py:component_init()] - WLED: Error HTTP 599: Unknown
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/wled.py", line 144, in _send_wled_command
    response = await self.client.fetch(request)
tornado.curl_httpclient.CurlError: HTTP 599: Unknown

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/wled.py", line 292, in component_init
    await ret
  File "/home/pi/moonraker/moonraker/components/wled.py", line 106, in initialize
    await self.set_wled(self.initial_red,
  File "/home/pi/moonraker/moonraker/components/wled.py", line 193, in set_wled
    await self._send_wled_command(
  File "/home/pi/moonraker/moonraker/components/wled.py", line 155, in _send_wled_command
    raise self.server.error(msg)
utils.ServerError: WLED: Error HTTP 599: Unknown
2021-12-23 12:13:23,941 [power.py:_send_http_command()] - Error sending 'tasmota' command: info
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/power.py", line 390, in _send_http_command
    response = await self.client.fetch(url)
tornado.curl_httpclient.CurlError: HTTP 599: Unknown
2021-12-23 12:13:23,941 [power.py:refresh_status()] - Error Refeshing Device Status: printer
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/power.py", line 390, in _send_http_command
    response = await self.client.fetch(url)
tornado.curl_httpclient.CurlError: HTTP 599: Unknown

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/power.py", line 409, in refresh_status
    state = await self._send_status_request()
  File "/home/pi/moonraker/moonraker/components/power.py", line 679, in _send_status_request
    res = await self._send_tasmota_command("info")
  File "/home/pi/moonraker/moonraker/components/power.py", line 676, in _send_tasmota_command
    return await self._send_http_command(url, command)
  File "/home/pi/moonraker/moonraker/components/power.py", line 395, in _send_http_command
    raise self.server.error(msg)
utils.ServerError: Error sending 'tasmota' command: info
2021-12-23 12:13:23,941 [moonraker.py:_initialize_component()] - Component [power] failed post init
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/moonraker.py", line 219, in _initialize_component
    await ret
  File "/home/pi/moonraker/moonraker/components/power.py", line 111, in component_init
    await ret
  File "/home/pi/moonraker/moonraker/components/power.py", line 383, in initialize
    await self.refresh_status()
  File "/home/pi/moonraker/moonraker/components/power.py", line 414, in refresh_status
    raise self.server.error(msg) from None
utils.ServerError: Error Refeshing Device Status: printer
2021-12-23 12:13:23,942 [moonraker.py:add_warning()] - Component 'power' failed to load with error: Error Refeshing Device Status: printer
```

Arksine, 2021-12-24:
This appears to be some kind of issue with Tornado's pycurl based http client.  I notice that you are running Debian Bullseye, which ships with Python 3.9.  Pycurl offically only supports up to python 3.8, so its possible that there is some kind of underlying conflict resulting an an error when Moonraker attempts to queue multiple client requests.

From Moonraker's perspect, the longterm solution is like to move away from Tornado's http framework to something else.  This isn't something I can get accomplished quickly, it will take time to switch frameworks and test before I can get it merged.  In the meantime you have a few options:
- Remove wled and see if that helps the issue
- Use MQTT to toggle your tasmota switch
- Try running on Debian Buster rather than Debian Bullseye

RcTomcat1, 2022-01-23:
Same Error regarding power after a cold boot with a shelly plug. Restarting Moonraker clears it in my case.

Config:
[power shelly_plug]
type = shelly
address = 192.168.3.16
user = myuser
password = mypass

"Error Refeshing Device Status: shelly_plug" and "tornado.curl_httpclient.CurlError: HTTP 599: Unknown"
Log looks identical to the one posted by the TE in that regard.

I am running  a current raspbian which is debian bullseye based as far as i know.

QuiNz0r, 2022-01-27:
same issue, when I boot I have to restart moonraker then it can connect, however: triggering my tasmota at shutdown doesn't work unfortunately. Can't setup an MQTT Broker in my environment, any fix?

[image]

Arksine, 2022-01-28:
I tried adding some retries to the http request, sadly it appears to be a dependency issue so I'm not sure if this will help.  You can give it a try though.

I'm not clear on what you mean by "triggering my tasmota at shutdown doesn't work".  Klippy shutdown?  This could be something unrelated, I can provide better feedback if you attach `moonraker.log` showing the issue.

Also, if anyone is getting 599 errors it might be useful to set `enable_debug_logging: True` in the `[server]` section.  This will produce large logs, so make sure you disable it after you have reproduced the error.

RcTomcat1, 2022-01-28:
I just updated to v0.7.1-306. Things looked ok until i did a cold boot.
Now i can't get a connection via the webinterface. Even minutes after the pi is booted up there is no webinterface available.
[image]

Moonraker.log seems to not being updated anymore. Last entry is several reboots in the past. Can't figure out why there is no log anymore. Commit 04477705b07f2fefd7666c7d3159d5e9ee17ee47 breaks moonraker for me.

I used git reset --hard e103accdfbe18e096e5574a25e6cb99e7678f3fe  to revert to the build before.
I worked my way through the commits and your change for the http requests with commit 1cac7399f918a05656a14582a5a891fdac69b1b1 still shows the power error after a cold boot. (at least it works)

Log attached

[moonraker.log](https://github.com/Arksine/moonraker/files/7957557/moonraker.log)

Arksine, 2022-01-28:
@RcTomcat1 The log is showing a dirty repo and a high number of unsafe shutdowns.  Are you removing power from the Pi without doing a graceful shutdown?  This can corrupt the filesystem, the git repo, and also the database.

Its clear that the `power.py` module has been corrupted or edited, as its reporting that it doesn't have a `load_component` method.  If Moonraker isn't logging this likely means its unable to start, so its possible that a critical file got modified with the result being a dead service.

You can try to reclone the repo:
```
sudo systemctl stop moonraker
rm -rf ~/moonraker
git clone https://github.com/Arksine/moonraker.git
sudo systemctl start moonraker
```

FWIW, if the filesystem is corrupted then you will have to reflash the OS.  Its possible that the Pi can recover the FS automatically with fsck though

RcTomcat1, 2022-01-28:
Thanks. That is quite interesting as the log is newly made just for the purpose of recreating the issue.
I just recloned the repo to be safe.
This log is done after a sudo shutdown now, so it should be a clean boot.

The log clearly shows the issue with the power.py.
Interesting thing is that after doing another restart of moonraker, the error seems to be gone and so is the message in the gui.
[moonraker.log](https://github.com/Arksine/moonraker/files/7958190/moonraker.log)

Arksine, 2022-01-28:
Thanks.  The debug logging helps here because curl outputs something useful.  It says that the network is unreachable.  The service is supposed the start after the network is available.

I'm going to add some delay between retries, however I need to limit how long moonraker stays down between restarts.  We need systemd to do the right thing here and launch moonraker after `network-online.target`.  Just in case, it might be useful to validate that the moonraker service file is correct.

If you run:
```
cat /etc/systemd/system/moonraker.service
```
you should get ouput that looks like the following:
```
#Systemd service file for moonraker
[Unit]
Description=API Server for Klipper
Requires=network-online.target
After=network-online.target

[Install]
WantedBy=multi-user.target

[Service]
Type=simple
User=pi
RemainAfterExit=yes
WorkingDirectory=/home/pi/moonraker
ExecStart=/home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/moonraker.py -c /home/pi/printer_config/moonraker.conf -l /home/pi/logs/moonraker.log
Restart=always
RestartSec=10
```
In particular you should look at the `[Unit]` section and validate that the `Requires` and `After` fields are present.

RcTomcat1, 2022-01-28:
Mine looks like this:
```#Systemd service file for moonraker
[Unit]
Description=Starts Moonraker  on startup
After=network.target

[Install]
WantedBy=multi-user.target

[Service]
Type=simple
User=pi
RemainAfterExit=yes
ExecStart=/home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/moonraker.py -l /home/pi/klipper_logs/moonraker.log -c /home/pi/klipper_config/moonraker.conf
Restart=always
RestartSec=10
```

Altered it so it reflects the required and after fields like in your post.
Seems to do the trick, no more error message after a cold boot.
Thank you so much

QuiNz0r, 2022-01-29:
everything works perfectly now for me, big thx!

Arksine, 2022-01-29:
Please check your service file as indicated above.  Note that if you aren't using Moonraker's `install-moonraker.sh` to do the installation your service file may not be correctly configured.

Arksine, 2022-03-16:
This issue should be resolved with the the service script, so I'm going to close this.

closed 2022-03-16 by Arksine

janmaier2277, 2022-05-12:
The error persists for me, even with 

`#Systemd service file for moonraker
[Unit]
Description=Starts Moonraker 1 on startup
Requires=network-online.target
After=network-online.target`

QuiNz0r, 2022-05-12:
> The error persists for me, even with
> 
> > #Systemd service file for moonraker
> > [Unit]
> > Description=Starts Moonraker 1 on startup
> > Requires=network-online.target
> > After=network-online.target

https://github.com/Arksine/moonraker/issues/317#issuecomment-1024129192

fixed it for me

janmaier2277, 2022-05-12:
> [#317 (comment)](https://github.com/Arksine/moonraker/issues/317#issuecomment-1024129192)
> 
> fixed it for me

Yeah i dont know, seems like network-online.target is reached before i get network connectivity. Delaying the start up by a few seconds fixes it for me. Am using LAN, DHCP & running Debian Bullseye on x86_64.

QuiNz0r, 2022-05-12:
> Yeah i dont know, seems like network-online.target is reached before i get network connectivity. Delaying the start up by a few seconds fixes it for me. Am using LAN, DHCP & running Debian Bullseye [...]

Which Method for controlling your [power] are you using? Accessing tasmota via https:// works for me but mqtt still gives me this error so maybe try another method if possible

hkasap83, 2022-05-18:
> Yeah i dont know, seems like network-online.target is reached before i get network connectivity. Delaying the start up by a few seconds fixes it for me. Am using LAN, DHCP & running Debian Bullseye [...]

which line did you change to delay the startup? Is it "RestartSec=10"?

janmaier2277, 2022-05-19:
Im using tasmota. The error will persist since network-online.target seems to be not working on debian bullseye as intended. 

RestartSec probably wont work, since moonraker does start and just unloads the power module. I added this to my moonraker.service which will make the service wait for the specified server to be pingable:

hkasap83, 2022-05-19:
> Im using tasmota. The error will persist since network-online.target seems to be not working on debian bullseye as intended.
> 
> RestartSec probably wont work, since moonraker does start and just unloads the power module. I added this to my moonraker.service which will make the service wait for the specified server (im pinging my modem) to be reachable. It is a rather dirty workaround, but i couldnt get network-online.target working consistently: ExecStartPre=/bin/sh -c 'until ping -c1 192.168.178.1; do sleep 1; done;'
> 
> ExecStartPre=/bin/sleep 5 also works, just make sure you set the value high enough for network connectivity.

The error persists for me. I tried lots of combination even with your advice but i still have to restart moonraaker.conf file from the fluidd web dashboard or restart the moonraker service by using putty (sudo service moonraker restart) to activate the smart plug (homeassistant). Can you share your "cat /etc/systemd/system/moonraker.service" output.

QuiNz0r, 2022-05-20:
Ok I have the same error again too, I switched from MQTT to Home assistant power switch and getting the exact same issue again :(

QuiNz0r, 2022-05-20:
[moonraker (7).LOG](https://github.com/Arksine/moonraker/files/8744825/moonraker.7.LOG)

Arksine, 2022-07-01:
FYI, I made some changes to how Moonraker initializes power devices since Bullseye doesn't seem to fully start the network before Moonraker starts.  Moonraker will now attempt to initialize indefinitely, thus eliminating the warning upon failure.

The downside to this change is that if the user will have to use the log to determine if the device is misconfigured since no warning is generated.

QuiNz0r, 2022-07-01:
> FYI, I made some changes to how Moonraker initializes power devices since Bullseye doesn't seem to fully start the network before Moonraker starts. Moonraker will now attempt to initialize [...]

thaaank you, this issue was driving me crazy