Arksine/moonraker #318: MQTT Power Plug

nicedevil007, 2021-12-28:
### What happened

Configure my MQTT Powerplug like it is written in the Documentation (just it is now a Shelly Plug instead of tasmota one)

### What did you expect to happen

No errors in Mainsail about wrong configuration

### How to reproduce

Add these lines to moonraker.conf

```
[power Shelly]
type: mqtt
#qos:
command_topic: shellies/Abstellkammer_Drucker/relay/0/command
command_payload:
    {% if command == "on" %}
      on
    {% else %}
      off
state_topic: shellies/Abstellkammer_Drucker/relay/0
#query_topic: shellies/Abstellkammer_Drucker/command
#query_payload: update
#query_after_command: True
off_when_shutdown: False
on_when_upload_queued: True
locked_while_printing: True
restart_klipper_when_powered: False
```

### Additional information

Output of the error:

```
Unparsed config option 'command_payload: {% if command == "on" %} on {% else %} off' detected in section [power Shelly]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
```

Arksine, 2021-12-28:
You are missing an `{% endif %}` in your template.  That said, you don't need the if statement at all if your payload is simply "on" or "off", you can simply send the command:
```
command_payload: {command}
```

nicedevil007, 2021-12-28:
> You are missing an `{% endif %}` in your template. That said, you don't need the if statement at all if your payload is simply "on" or "off", you can simply send the command:
> 
> ```
> command_payload: {command}
> ```
> 
> Edit: I see there is an error in the documentation. I'll fix that.

OK it is working now :)
Small open question at this point - how is the restart of Klipper handled now after I turned the plug on?

On the old version with a "normal" shelly setup there was the option:

`restart_delay: 10`

That seems to not work anymore.

```
Unparsed config option 'restart_delay: 10' detected in section [power Shelly]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
```

Arksine, 2021-12-29:
For the `restart_delay` option to be available you need to set `restart_klipper_when_powered: True` in the configuration.  This behavior hasn't changed.

nicedevil007, 2021-12-29:
Ty.... I dont know why I switched it to false. Working great now!

closed 2021-12-29 by nicedevil007