Arksine/moonraker #321: klipper wont connect to printer moon raker error

metric1578, 2021-12-30:
### What happened

can't seem to connect to printer with klipper/moonraker. attached is a klippy log:

{"error": {"code": 404, "message": "Not Found", "traceback": "Traceback (most recent call last):\n\n  File \"/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py\", line 1704, in _execute\n    result = await result\n\n  File \"/home/pi/moonraker/moonraker/app.py\", line 639, in get\n    self.root, absolute_path)\n\n  File \"/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py\", line 2778, in validate_absolute_path\n    raise HTTPError(404)\n\ntornado.web.HTTPError: HTTP 404: Not Found\n"}}

### What did you expect to happen

I expected to connect to printer via fluidd. I tried updating rpi4 but didn't seem to help. I can log on fluidd site and see my config files but it just won't connect to printer.

### How to reproduce

installed latest release fluidd,klipper,moonraker and updated.

Arksine, 2021-12-30:
You will need to supply `moonraker.log` and `klippy.log`.  My suspicion is that you don't have the unix socket enabled.

Arksine, 2022-01-06:
As discussion seems to have concluded I'm going to close this.  If you need any further assistance I suggest trying the discord.

closed 2022-01-06 by Arksine

gknops, 2023-09-23:
I am seeing the same issue, was there any discussion with possible hints?

yw4z, 2024-06-22:
same error appeared for me after updating klipper from fluidd UI. i guess this method changed an environment variable

[moonraker.log](https://github.com/user-attachments/files/15936524/moonraker.log)

probably i will uninstall and reinstall klipper to fix it.  but why :)

Romano26, 2025-06-09:
I used kiauh to restore the previous version of klipper