Arksine/moonraker #323: No moonraker connection after update

INeDiA, 2021-12-31:
### What happened

Updated to v0.7.1-280 via KIAUH script, suddenly there is no connection to moonraker.
Fluidd asks me to add a new printer, which of course it can't find. Both klippy and moonraker logs look ok:
[klippy.log](https://github.com/Arksine/moonraker/files/7796205/klippy.log)
[moonraker.log](https://github.com/Arksine/moonraker/files/7796206/moonraker.log)

My moonraker config: [moonraker.txt](https://github.com/Arksine/moonraker/files/7796216/moonraker.txt)

Moonraker service is up and running
[image]

 
I also had a telegram bot running which now keeps smapping me with:
```
Bot online, no moonraker connection!
Connection failed. 
Failing...
```

### What did you expect to happen

Connect to moonraker

### How to reproduce

Can't reproduce

Arksine, 2021-12-31:
I don't see any attempt from a client to connect in moonraker.log.  The 404 you posed from the console log should also show up in moonraker.log.  Are you running multiple instances?  Which endpoint did you attempt to access that resulted in the 404?

Also, the klippy.log you uploaded isn't the full log.

Arksine, 2021-12-31:
Looking further, something is wrong with the connection to Klippy.  Moonraker hasn't received its list of endpoints.  So moonraker can connect to the unix socket but it doesn't seem to be receiving any response.  Get Klippy fixed and things should work properly.

INeDiA, 2021-12-31:
Thanks for the fast reply!

Re-exported the log after the 404 hit, I can non see the 404 get. Here is the [full log](https://paste.c-net.org/ReceiptsSellout).
No, I do not have multiple instances.
Klippy.log is quite verbose, here is [the full thing](http://paste.c-net.org/CoursingBenefit).

If I may, where do you see something wrong with klipper? I tried to remove it/reinstall it but I can't see any changes.
Thanks!

Arksine, 2021-12-31:
The full logs for both Klipper and Moonraker look ok.  I see the 404s,  something is trying to access the instance directly without specifying and endpoint.  For example, `http://192.168.100.100:7125/` will always return a 404 because moonraker does not serve static files and it doesn't reference an http API, however `http://192.168.100.100:7125/server/info` should return a json response.

As for telegram/fluidd, I see attempts from fluidd to connect.  Something appears to be going wrong on its end.  I can't reproduce the issue here though.  Id don't use telegram, but that error doesn't provide enough context for me to comment on it.  I would assume that its installed on the host alongside Moonraker, though I see no attempts to connect from the 127.0.0.1.

INeDiA, 2021-12-31:
Don't worry about Telegram, it's a minor issue, I was just adding info for troubleshooting but I now understand this was confusing.

Yes I can get a json response:
[image]

That websocket_count doesn't look good; does this help to address the issue?

Arksine, 2021-12-31:
It looks to me like Fluidd just isn't trying to connect to the websocket.  Maybe resetting the layout or clearing the cache will fix it.

Arksine, 2021-12-31:
Also give mainsail a try.  If neither connects then there may be some sort an issue with the Moonraker installation.

INeDiA, 2021-12-31:
> It looks to me like Fluidd just isn't trying to connect to the websocket. Maybe resetting the layout or clearing the cache will fix it.

I'm pretty sure Moonraker is having some issue; neither fluidd nor mainsail (nor the telegram bot) can connect to it, same issue.
I tried with multiple moonraker instances, just to be sure that the port was free: same result.
I tried to rebuild the environment (./install-moonraker.sh -r), no luck.
I tried to remove every single moonrake related file and folder, then reinstall, still same result.

Isn't that websocket_count supposed to be > 0?

Arksine, 2021-12-31:
The websocket count is only above 0 if a websocket is connected.  In the logs you showed me Fluidd wasn't attempting to open a websocket.  However if Mainsail won't open one either then that does suggest an issue with Moonraker.  Does the developer console report any errors when attempting to connect in Fluidd or Mainsail?  Moonraker logs all attempts at opening a websocket.  Nothing has changed recently with the way websockets are handled, so I am unsure as to why an update would have resulted in difference.

INeDiA, 2022-01-01:
I'm sorry, what developer console? Do you mean the built-in browser one?
I have some json response so something is answering I guess:
[image]

Yes I'm adding the port, of course!
About the shutdowns: yes it's totally my fault, the pi is powered directly from the printer 😅

Arksine, 2022-01-01:
Correct, the developer console built in to the browser.  For example, this one from Chrome that shows the websocket request opened by fluidd:
[image]

With Chrome you need to open the console first before navigating.  If screenshot you posted is all that is showing up then it matches what the log says, your browser isn't making a request to open a websocket.

>About the shutdowns: yes it's totally my fault, the pi is powered directly from the printer sweat_smile

I would recommend against this.  It might not be the source of this particular issue but it will likely cause a problem at some point.

INeDiA, 2022-01-01:
In the end I gave up and just wiped the whole thing.
I was back online in 30' without any issue, so definetly something broke during the update or because of my reckless shutdown policy.

Anyway, thanks for your support, I really appreciate it!

closed 2022-01-01 by INeDiA