Arksine/moonraker #324: MQTT Setup for filament_switch_sensor

nicedevil007, 2022-01-02:
### What happened

No message will be published over MQTT for the `filament_switch_sensor`

### What did you expect to happen

A message if filament is detected or not will be published over MQTT

### How to reproduce

moonraker.conf

```
[mqtt]
address: mqtt.skynet
port: 1883
username: {secrets.mqtt_credentials.username}
password: {secrets.mqtt_credentials.password}
mqtt_protocol: v3.1.1
enable_moonraker_api: True
instance_name: sidewinder
status_objects:
    webhooks
    toolhead=position,print_time,homed_axes
    extruder=temperature,target
    heater_bed=temperature,target
    fan=speed
    idle_timeout=state
    virtual_sdcard=progress,is_active,file_position
    print_stats=state,messge,total_duration,print_duration,filename
    filament_switch_sensor Filament_Sensor=filament_detected
default_qos: 0
#   The default QOS level used when publishing or subscribing to topics.
#   Must be an integer value from 0 to 2.  The default is 0.
#api_qos:
#   The QOS level to use for the API topics. If not provided, the
#   value specified by "default_qos" will be used.
```

printer.cfg
```
[filament_motion_sensor Filament_Sensor]
detection_length: 7.0
extruder: extruder
switch_pin: PG11
pause_on_runout: True
event_delay: 10.0
pause_delay: 0.5
runout_gcode:
    # keep in mind that this requires pause_on_runout = True
    M118 Filament runout detected, retracting
    G0 E10 F300                 # extrude filament to get better blob on end
    G0 E-100 F700               # retract additional filament to move out of melt zone
    G92 E0
    M118 Insert new filament now
```

### Additional information

I guess the main issue here will be the " " between `filament_switch_sensor` and `Filament_Sensor`

Arksine, 2022-01-02:
The space isn't an issue, its should work fine.  You have `filament_switch_sensor` in your [mqtt] configuration but `filament_motion_sensor` in printer.cfg.

nicedevil007, 2022-01-02:
wtf... missed that :/ my bad! and ty again!

closed 2022-01-02 by nicedevil007