Arksine/moonraker #331: moonraker.service can't up in Orangepi zero board

sullenyang, 2022-01-14:
### What happened

1. write SD card with image: Armbian_21.08.1_Orangepizero_focal_current_5.10.60.img.xz  or Orangepizero_2.0.8_ubuntu_focal_server_linux5.4.65.7z
2. create new user with password
3. sudo apt-get update / sudo apt-get upgrade
4. git clone https://github.com/th33xitus/kiauh.git
5. RUN ./kiauh/kiauh.sh
6. install klipper moonraker and fluidd
7. use 192.168.1.186 to access fluidd, can't connect to moonraker
8. check service in SSH, moonraker.service show error:   "Active: activating"  "Main PID: 1773 (code=exited, status=1/FAILURE)"
9. this moonraker version is v0.7.1-299. 
10. then copy an old version (v0.7.1-273) file to overwrite it, it works.  

### What did you expect to happen

moonraker.service can up normally as before version.

### How to reproduce

1. write SD card with image: Armbian_21.08.1_Orangepizero_focal_current_5.10.60.img.xz  or Orangepizero_2.0.8_ubuntu_focal_server_linux5.4.65.7z
2. create new user with password
3. sudo apt-get update / sudo apt-get upgrade
4. git clone https://github.com/th33xitus/kiauh.git
5. RUN ./kiauh/kiauh.sh
6. install klipper moonraker and fluidd
7. use 192.168.1.186 to access fluidd, can't connect to moonraker
8. check service in SSH, moonraker.service show error:   "Active: activating"  "Main PID: 1773 (code=exited, status=1/FAILURE)"
9. this moonraker version is v0.7.1-299. 
10. then copy an old version (v0.7.1-273) file to overwrite it, it works.

Arksine, 2022-01-14:
KIAUH is another project, I can't provide assistance for installations that use it.  My suggestion would be to seek help on the Klipper discord.

My guess is that you are missing a dependency, there should be a traceback when you check the service status in SSH giving more information.

Arksine, 2022-03-16:
Since this issue has become inactive I am going to close this.  As mentioned above, it seems likely that a dependency was not installed when you updated.

closed 2022-03-16 by Arksine