Arksine/moonraker #333: no moonraker connection. please check moonraker status and / or refresh

alessandro111297, 2022-01-16:
### What happened

After Fluidd update in Fluidd API, there is this error "no moonraker connection. please check moonraker status and / or refresh"
[image]
This is the log file:
[fluidd-error.log.txt](https://github.com/Arksine/moonraker/files/7878098/fluidd-error.log.txt)
[moonraker.log.log](https://github.com/Arksine/moonraker/files/7878099/moonraker.log.log)
After this I also updated moonraker and fluidd from kiuah 

### What did you expect to happen

.

### How to reproduce

.

Arksine, 2022-01-17:
Presumably Moonraker failed to start after update, its possible that one or more new dependencies failed to install.
Run the following command and let me know what the output is:

```
sudo journalctl -xe -u moonraker.service
```

If there is traceback in there copy it over.  The other possibility is repo corruption, however I think that is unlikely.

alessandro111297, 2022-01-17:
Solved,  installed dependecies with the command: ~/moonraker/scripts/install-moonraker.sh and finally work.
Thank you very much

closed 2022-01-17 by alessandro111297

Dev-Mug, 2022-02-09:
> Presumably Moonraker failed to start after update, its possible that one or more new dependencies failed to install. Run the following command and let me know what the output is:
> 
> ```
> sudo journalctl -xe -u moonraker.service
> ```
> 
> If there is traceback in there copy it over. The other possibility is repo corruption, however I think that is unlikely.
> 
> Edit:
> 
> You could also try running the install script to make sure dependencies are updated:
> 
> ```
> ~/moonraker/scripts/install-moonraker.sh
> ```

This worked for me as well, Had the same issue. Thanks guys!

d3dorsett, 2022-04-22:
I used kiauh to install klipper with fluidd. After a recent update, I got this error too. Running the script above also fixed my issue.

It appears these files did not get updated...
Downloading https://www.piwheels.org/simple/certifi/certifi-2021.10.8-py2.py3-none-any.whl (151 kB)
Downloading https://www.piwheels.org/simple/charset-normalizer/charset_normalizer-2.0.12-py3-none-any.whl (44 kB)
Downloading https://www.piwheels.org/simple/urllib3/urllib3-1.26.9-py2.py3-none-any.whl (155 kB)
Downloading https://www.piwheels.org/simple/requests/requests-2.27.1-py2.py3-none-any.whl (63 kB)

anderfrank, 2022-11-14:
> Presumably Moonraker failed to start after update, its possible that one or more new dependencies failed to install. Run the following command and let me know what the output is:
>
> ```
> sudo [...]

Thanks for this, helped fix my install after I had to rebuild.

JamieKaybg, 2023-01-22:
> I had a similar problem, but solved by deleting the moonraker.conf file, uninstalling Moonraker and installing again.

I believe this will also put me back on track to resolving a bad gate way in moonraker.cfg, after changing the host server up address. My delima is I'm unsure how to get CLI in Moonraker seems this was a distribution from creality Sonic pad, installed on a Windows system. Any recommendations greatly appreciated. Is it possible to access CLI from Sonic pad

Moxie15, 2024-09-06:
Where do I put the command?????

clmcgrath, 2024-11-06:
for anyone coming here for the bad gateway error check your kllippy logs ,  i had an issue today where i uninstalled obico like 2 weeks ago but it didnt remove the include from moonraker moonraker hadnt restarted till today when i needed to reboot my pi moonraker refused to start and gave bad gateway errors i checked my log file and i was getting missing include file errors , checked my moon raker conf and the was an old include 
moral of the story always check your klippy logs if you are having errors

clmcgrath, 2024-11-06:
> Where do I put the command?????

ssh into your printer to execute any non gcode specific command 
i would check your klippy logs forst tho 
which can be reached by ssh  if your moonraker isn't starting ==> cat ~/printer_data/logs/klippy.log

zemlyanskii-artem, 2025-01-12:
вот такая же проблема . Не могу решить ! непонимаю куда вставлять ваши команды!

lol12433, 2025-02-23:
i also have the same issue but on the home screen

Joscal76, 2025-05-16:
Alguien puede ayudarme con el acceso al cambio para solucionar el mismo problema? 
Gracias de antemano

lol12433, 2025-05-16:
to  joscal76: conseguir un nuevo chip EMC para la impresora.