Arksine/moonraker #337: When uploading a large gcode file (>400MB), the CPU load will be too high, causing the moonraker service to restart

kluoyun, 2022-01-19:
### What happened

When I test uploading a large file, the MoonRaker service occupies more than 100% of the CPU, and the moonraker service restarts, resulting in file loss.
How does MoonRaker handle files?

### What did you expect to happen

Complete upload of large files

### How to reproduce

Upload a GCODE larger than 400mb in Fluidd

Arksine, 2022-01-20:
Moonraker streams uploads to a temporary path, then moves the file to the desired folder.  Moonraker can't use more than 100% CPU,  most of what you would see there is Moonraker waiting for I/O.  If the service is restarting I need you to provide the `moonraker.log` showing this behavior.  This could very well be a system issue.

kluoyun, 2022-01-20:
I did the test again yesterday and found that Fluidd would have a page reload problem, and Mainsail everything was fine. moonraker.log is also normal

Arksine, 2022-01-20:
It sounds like the issue is with Fluidd, not Moonraker.

kluoyun, 2022-01-20:
Yep, I got it wrong earlier and thought it was Moonraker's problem. Feel sorry

Arksine, 2022-01-20:
Not a problem. I hope they are able to get it resolved.

kluoyun, 2022-01-20:
Yes

closed 2022-01-21 by kluoyun