Arksine/moonraker #338: "Websocket closed while writing"

vellocet69, 2022-01-19:
### What happened

The web interface randomly stops being accessible, happens several times per print session. Will last for a few minutes, then it'll be reachable again. During that timeframe, the Pi is still connected to the home network and the print is unaffected as well. Moonraker log shows something like this:

`[websockets.py:_process_messages()] - Websocket closed while writing: 1234567890
 [websockets.py:remove_websocket()] - Websocket Removed: 1234567890`

After a while:

`[app.py:log_request()] - 101 GET /websocket?token=XYZ...`
`[websockets.py:add_websocket()] - New Websocket Added: 2345678910`

...and the web interface will be reachable again.

### What did you expect to happen

There should be no interruptions of connectivity as long as a network connection is established. 

### How to reproduce

Anytime I'm connected via browser, time until the issue appears is random - might be 2 minutes, might be half an hour. Doesn't matter if I'm actively doing something or if it's just idling.

Arksine, 2022-01-19:
Most likely the client is closing the Websocket.  I would recommend testing another to Frontend to see if you can reproduce the issue.  If we can determine that Moonraker is the cause then I will investigate further.

throwmeawayafterthat, 2022-01-20:
So you think Fluidd might be the issue and I could just try Mainsail?

Arksine, 2022-01-20:
That would be my first recommendation.  If the issue persists with Mainsail we can investigate further.  Otherwise you can attempt to create a ticket with Fluidd.

BuddyBing, 2022-01-20:
I'm also getting similar issues using Mainsail so I'm not sure it is frontend specific.  Here are the errors pulled out of my log:
2022-01-17 00:06:24,817 [websockets.py:_process_messages()] - Websocket closed while writing: 1719187968 
2022-01-17 00:06:29,789 [websockets.py:remove_websocket()] - Websocket Removed: 1719187968 
2022-01-17 00:58:12,817 [authorization.py:_prune_conn_handler()] - Trusted Connection Expired, IP: 192.168.7.191

throwmeawayafterthat, 2022-01-20:
Here's a full moonraker log right after regaining connectivity from a few minutes ago.
[Pastebin link](https://pastebin.com/jVK8B1yg)

Unfortunately, I have to rely on wireless atm. Browser for me is Firefox on my desktop and DuckDuckGo on my phone.

BuddyBing, 2022-01-20:
@Arksine I think @throwmeawayafterthat  and I may have two separate issues (so let me know if you need me to create another issue log).  Here is my moonraker log which shows a certificate issue.  This appears to only be happening on my Raspberry PI Zero 2 (as I have other pi's that have been working fine without cert issues). I am using Chrome, but tried safari as well.  I am also using a eero mesh wifi network.

[moonraker.log.2022-01-12.zip](https://github.com/Arksine/moonraker/files/7907255/moonraker.log.2022-01-12.zip)

Arksine, 2022-01-20:
Ok, I'm gonna take a look at these logs.  FYI, I just pushed a commit that adds some additional logging when a websocket is closed.  Specifically it logs the close code, reason, and time elapsed since the last pong received.  This can help us determine if the client is requesting a closed socket.  If we're lucky it will log one of the error codes.

Arksine, 2022-01-20:
@throwmeawayafterthat
After looking at your log I see three disconnections:
1) After a Moonraker update due to a Server Restart
2) After a request to reboot (this is where you see the `Websocket closed while writing: 1719187968` message)
3) Finally several minutes after the update manager refreshes.

If Fluidd is giving you several "Lost Moonraker Connection" messages my guess is that there is an issue client side, particularly if the websocket is remaining open.  If you notice this, grab the log and upload again.  The new logging may help here, however if the websocket isn't closing.

@BuddyBing 
Regarding the certificate issue, if you are referring to this line:
```
2022-01-12 20:47:34,920 [app.py:listen()] - SSL Certificate/Key not configured, aborting HTTPS Server startup
```
That is normal.  Moonraker wont start HTTPS without a Cert configured, but HTTP starts just fine.

With regard to the websocket issue, I do see where the websocket opens and closes repeatedly.  I have one theory about the cause, its possible that the client is receiving a notification before its ready and its either closing or crashing the websocket immediately.  The new logging I added should help us narrow it down.

throwmeawayafterthat, 2022-01-20:
I messed around quite a bit during that session, did several updates and such. I just updated moonraker and wait for another disconnect now so I can get a new log.

throwmeawayafterthat, 2022-01-20:
Okay I'll copy the relevant parts of the log right here..."pong time elapsed". Have to add that this time I had it open on my desktop and my phone as well.

`2022-01-20 21:15:01,377 [shell_command.py:_check_proc_success()] - Command (/home/pi/moonraker-env/bin/python /home/pi/moonraker/moonraker/components/file_manager/metadata.py -p /home/pi/gcode_files -f "xyzCalibration_cube_20m_0.2mm_220C_PET_ENDER3V2.gcode") successfully finished`
`2022-01-20 21:15:01,407 [app.py:log_request()] - 201 POST /server/files/upload (192.168.178.26) [_TRUSTED_USER_] 719.42ms`
`2022-01-20 21:26:04,134 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 192.168.178.29`
`2022-01-20 21:26:04,912 [app.py:log_request()] - 101 GET /websocket?token=PRKGP5OKSBJAIQ5BOLFX5Z46A2A5HL4R (192.168.178.29) [_TRUSTED_USER_] 6.76ms`
`2022-01-20 21:26:04,913 [websockets.py:open()] - Websocket Opened: ID: 1731494256, User Agent: Mozilla/5.0 (Linux; Android 11) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/97.0.4692.87 Mobile DuckDuckGo/5 Safari/537.36`
`2022-01-20 21:27:58,792 [websockets.py:on_close()] - Websocket Closed: ID: 1731494256 Close Code: None, Close Reason: None, Pong Time Elapsed: 3.80`
`2022-01-20 21:45:03,120 [app.py:log_request()] - 101 GET /websocket?token=5JOQDELQSJA6U3WV6ALTQHG2A7Z7LOFP (192.168.178.29) [_TRUSTED_USER_] 4.64ms`
`2022-01-20 21:45:03,121 [websockets.py:open()] - Websocket Opened: ID: 1731587312, User Agent: Mozilla/5.0 (Linux; Android 11) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/97.0.4692.87 Mobile DuckDuckGo/5 Safari/537.36`
`2022-01-20 21:45:07,483 [websockets.py:on_close()] - Websocket Closed: ID: 1731587312 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 4.37`
`2022-01-20 21:45:09,257 [app.py:log_request()] - 101 GET /websocket?token=TFGAOOEQS2D5YFASUZFQLF4NPUVX3MSR (192.168.178.29) [_TRUSTED_USER_] 5.95ms`
`2022-01-20 21:45:09,258 [websockets.py:open()] - Websocket Opened: ID: 1730125136, User Agent: Mozilla/5.0 (Linux; Android 11) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/97.0.4692.87 Mobile DuckDuckGo/5 Safari/537.36`
`2022-01-20 21:47:28,521 [websockets.py:_process_messages()] - Websocket closed while writing: 1731586320`
`2022-01-20 21:47:29,455 [websockets.py:_process_messages()] - Websocket closed while writing: 1730125136`
`2022-01-20 21:47:33,491 [websockets.py:on_close()] - Websocket Closed: ID: 1731586320 Close Code: None, Close Reason: None, Pong Time Elapsed: 45.00`
`2022-01-20 21:47:34,263 [websockets.py:on_close()] - Websocket Closed: ID: 1730125136 Close Code: None, Close Reason: None, Pong Time Elapsed: 45.00`
`2022-01-20 21:48:20,994 [app.py:log_request()] - 101 GET /websocket?token=W3ZV7FIDVTEJHMXTWLJYHH35GQMK27CD (192.168.178.26) [_TRUSTED_USER_] 3.01ms`
`2022-01-20 21:48:20,995 [websockets.py:open()] - Websocket Opened: ID: 1730124432, User Agent: Mozilla/5.0 (X11; Linux x86_64; rv:96.0) Gecko/20100101 Firefox/96.0`
`2022-01-20 21:49:25,936 [app.py:log_request()] - 101 GET /websocket?token=LTWTZGL74KHWECHINP5RY3OT3G2IO3R5 (192.168.178.29) [_TRUSTED_USER_] 5.22ms`
`2022-01-20 21:49:25,937 [websockets.py:open()] - Websocket Opened: ID: 1731496080, User Agent: Mozilla/5.0 (Linux; Android 11) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/97.0.4692.87 Mobile DuckDuckGo/5 Safari/537.36`

Arksine, 2022-01-21:
@throwmeawayafterthat
From that snippet it appears that the clients have become unresponsive.  Pinging is part of the websocket spec, when Moonraker sends a ping it should receive a pong back.  Moonraker pings every 10 seconds and expects at least one pong every 30 seconds.  Its not receiving a response, so the websocket is closed. 

I can think of a few possibilities that would cause this:
1) The websocket is being proxied and the upstream proxy is causing an issue.  I pushed another commit that logs whether or not a reverse proxy and/or load balancer has been detected
2) Wireless networking issue as mentioned previously
3) Client/Browser issue. Given the close timing of the two disconnections, its possible that Fluidd is receiving some response or notification that it can't handle.  FWIW, Fluidd is still searching for an active maintainer.
4) Something could be hanging up with Tornado's websocket implementation.  However if this is the case these issues should be more widespread and I should be able to reproduce locally.  That said, in the future I may move away from Tornado to another web framework, but this isn't something that can be done quickly.

throwmeawayafterthat, 2022-01-21:
I don't wanna be too early with my statement but it might actually be reason number 2. Today I disabled the onboard Wifi of my Pi 3b+ and used a TP Link WN722N v1.1 USB dongle instead. I didn't have a single disconnect since then for the last hour. I'll keep an eye on it.

Arksine, 2022-03-16:
It seems that this was a networking issue, so I'm going to close this.  Thanks.