Arksine/moonraker #340: Webcam isn't stable // uses a lot of CPU//crashes system

nickeeey, 2022-01-22:
### What happened

Hi, i use a c270 webcam with Moonraker and a Raspberry Pi 3b+. The Webcam worked perfectly at octoprint, but won't work right with Moonraker. I see the picture/ Video after startup but when i switch 2-3 times to the dashboard it just loads. Also my cpu power is going up and i can't access files from the server. I tried different resoultions and framerates

### What did you expect to happen

-

### How to reproduce

C270 Webcam on Moonraker with Pi 3b+

### Additional information

-

Arksine, 2022-01-22:
Moonraker doesn't serve the webcam stream, your issue is either with the webcam streaming daemon (mjpegstreamer, ustreamer, etc) or you are experiencing some kind of webcam driver issue.

Presumably you are using either MainsailOS or FluiddPi.  If so, I'd suggest requesting help on one of their respective discords.

nickeeey, 2022-01-22:
Hi, Thanks and sorry. Then i Close it here
Yeah its MainSailOs

closed 2022-01-22 by nickeeey