Arksine/moonraker #343: Tasmota not turning off

nickeeey, 2022-01-23:
### What happened

Hi, my tasmota plug is in the power menu, but it doesnt turn off after the Print.

[power tasmota_plug]
type: tasmota
address: 192.168.0.112
off_when_shutdown: True
off_when_shutdown_delay: 15
on_when_upload_queued: True
locked_while_printing: True
restart_klipper_when_powered: False

### What did you expect to happen

that the printer turns off 15sec after finishing the print

### How to reproduce

see above

Arksine, 2022-01-28:
Moonraker doesn't have an option to turn off the printer after a print.  However, you could accomplish this in Klipper.  The documentation explains how to call `set_device_power` from printer, put that in a delayed gcode to be executed when your print ends and it should power down the printer.

Arksine, 2022-03-16:
I'm going to close this as it should be resolved.  The documentation explains how to call the shutdown remote method from Klipper, so it is just a matter of adding the correct macros/delayed gcodes to accomplish a shutdown after print.

closed 2022-03-16 by Arksine