Arksine/moonraker #347: moonraker keeps disconnecting/restarting

clf139, 2022-01-26:
### What happened

Moonraker keeps disconnecting/restarting every couple of minutes

### What did you expect to happen

I expect moonraker to stay connected 

### How to reproduce

Continues to occur even after the following troubleshooting steps:

- Updated: moonraker, klipper, timelapse, fluidd, and OS packages

Logs and screenshots attached
[image]
.
[moonraker (4).log](https://github.com/Arksine/moonraker/files/7941998/moonraker.4.log)
[klippy (3).log](https://github.com/Arksine/moonraker/files/7941996/klippy.3.log)
[fluidd-error.log](https://github.com/Arksine/moonraker/files/7942015/fluidd-error.log)

Arksine, 2022-01-26:
There aren't any disconnections in moonraker.log.  This could be an issue with using nginx, you could try having fluidd connect to Moonraker directly via port 7125.  Otherwise it could be a local networking issue, or an issue with Fluidd.

Arksine, 2022-03-16:
This issue has become inactive, so I am going to close it.  As mentioned, the issue could be nginx or a networking issue.

closed 2022-03-16 by Arksine