Arksine/moonraker #348: Support for custom bound_service

mikenorgate, 2022-01-26:
### Is your feature request related to a problem? Please describe

I have created a simple service to disable/enable the USB ports so that the printer isn't back powered via USB when it should be off. I would like to use this service in the power control section. As a workaround I have named by service `MoonCord` as I'm not using that one

```
[Unit]
Description=USB Power

[Service]
ExecStart=sudo uhubctl -l 2 -a 1
ExecStop=sudo uhubctl -l 2 -a 0
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target
```

### Describe the solution you'd like

Ability to specify any service in the `bound_service` option

### Describe alternatives you've considered

Have the option to either run a custom script on power on/off or potentially build this functionality into Moonraker

Arksine, 2022-01-28:
The issue we have here is trying to balance user configuration with potentially dangerous options.  Moonraker intentionally filters the services that can be managed and I don't think its wise to expand beyond that.

That said, I'm not against whitelisting a `uhubctl` service.

mikenorgate, 2022-03-18:
Sorry I somehow completely missed your response on this. I think the issue with whitelisting a `uhubctl` service is that there isn't a service as part of the the `uhubctl` installation, so you would be relying on the user manually creating that service.

Would it be a better option to allow specifying a shell command in the config instead of relying on a service, although this could be potentially dangerous

11chrisadams11, 2022-04-19:
I also have created a custom service that I would like to be able to restart from the power menu (being able to save and restart it after editing the config would also be great). I don't think allowing custom services is any more dangerous than allowing custom git repos. Maybe there could be an option to allow it with a warning?

Jckf, 2022-10-16:
This might be of use https://github.com/Arksine/moonraker/pull/525