Arksine/moonraker #350: printer powers off when klipper starts (using relays and macros)

bluepinner, 2022-01-29:
### What happened

I have in my config:

[power printer]
type: gpio
pin: !gpio14
off_when_shutdown: True
restart_klipper_when_powered: True
initial_state: Off

and it works but when I enable:

#[power led]
#type: gpio
#pin: gpio18
#off_when_shutdown: True
#initial_state: Off

The printer shuts down when klipper starts. I can have each working perfect by itself but not together. Am I doing something wrong? Log Provided

### Client

Mainsail, KlipperScreen

### Browser

Chrome, Firefox, Microsoft Edge

### How to reproduce

Both macros enabled at the same time. Power printer on, restart klipper and the relay for the printer shuts off.

### Additional information

[moonraker.log](https://github.com/Arksine/moonraker/files/7964151/moonraker.log)

Arksine, 2022-01-30:
Klippy appears to immediately enter the shutdown state after connecting to Moonraker, which is in turn powering off the printer.  Check `klippy.log` to find the cause of the shutdown.

Arksine, 2022-03-16:
This issue has become inactive so I am going to close it.  As mentioned above, Klipper was the source of the shutdown.

closed 2022-03-16 by Arksine