Arksine/moonraker #352: moonraker not starting after update

tommyl2210, 2022-01-30:
### What happened

After updating klipper, moonraker and system packages today, moonraker doesn't start anymore. Log file shows that dbus is not available. 

Tried to use "install-moonraker.sh" by
`pi@mainsailos:~/moonraker $ sudo sh ./scripts/install-moonraker.sh`

but it also fails with:
`###### Installing python virtual environment...
ERROR: Could not open requirements file: [Errno 2] No such file or directory: '//scripts/moonraker-requirements.txt'
`

I also added the two recommended [machine] section options:
`[machine]
provider: none

[update_manager]
enable_packagekit: False
`

Any idea to get it back up running?

### Client

Mainsail

### Browser

Firefox

### How to reproduce

see in description

wissencat1, 2022-01-30:
It has to do with that and with Pip.
#349
Sadly we have to wait for an update.

Arksine, 2022-01-30:
This isn't related to #349, its related to an upstream bug introduced by pip.  See #351.

tommyl2210, 2022-01-30:
> This isn't related to #349, its related to an upstream bug introduced by pip. See #351.
> 
> Edit: wait, you also can't run the install script with sudo. However you will still run into the issue with pip.
> 
> Also, your log is showing a duplicate section error. Make sure you don't have a section in the config twice. For example, if you have `[update_manager]` in the config, move the `enable_packagekit: False` to that section.

thanks @Arksine ! I tried without sudo as well and now removed the extra sections completely. So I guess I/we have to wait until the upstream problem is fixed?

ckvsoft, 2022-01-30:
how we can go back?

Arksine, 2022-01-30:
I'm searching for a workaround to get pip working.  If you haven't run `install-moonraker.sh -r` you can try revert Moonraker with the following commands:
```
cd ~/moonraker
git reset --hard 9911b5c7dd2b8a8962a61e2cf9e98fa8fb1725c9
```

tommyl2210, 2022-01-30:
> 

Thank you so much! This worked for me. I will follow the discussion here and update once pip is fixed.

Fredy12335, 2022-01-30:
> If you haven't run `install-moonraker.sh -r` you can try revert Moonraker with the following commands:
> 
> ```
> cd ~/moonraker
> git reset --hard 9911b5c7dd2b8a8962a61e2cf9e98fa8fb1725c9
> sudo service moonraker restart
> ```
> 
> If you ran the install script with the `-r` switch you deleted your old env and we can't get it back until we can reach pip. I'm checking to see if I can find a workaround for this.

In my case:
 cd ~/moonraker
~/moonraker $ git reset --hard 9911b5c7dd2b8a8962a61e2cf9e98fa8fb1725c9
HEAD is now at 9911b5c app: don't allow open redirects in '/server/redirect'

I dont remember run this before: "install-moonraker.sh -r"

tommyl2210, 2022-01-30:
> > If you haven't run `install-moonraker.sh -r` you can try revert Moonraker with the following commands:
> > ```
> > cd ~/moonraker
> > git reset --hard 9911b5c7dd2b8a8962a61e2cf9e98fa8fb1725c9
> > sudo service moonraker restart
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > If you ran the install script with the `-r` switch you deleted your old env and we can't get it back until we can reach pip. I'm checking to see if I can find a workaround for this.
> 
> In my case: cd ~/moonraker ~/moonraker $ git reset --hard [9911b5c](https://github.com/Arksine/moonraker/commit/9911b5c7dd2b8a8962a61e2cf9e98fa8fb1725c9) HEAD is now at [9911b5c](https://github.com/Arksine/moonraker/commit/9911b5c7dd2b8a8962a61e2cf9e98fa8fb1725c9) app: don't allow open redirects in '/server/redirect'
> 
> I dont remember run this before: "install-moonraker.sh -r"

Same with me, restart the pi and it might work. Worked for me.

turbotrix, 2022-01-30:
Thanx Arksine,temp fix worked for me

toby5419, 2022-01-30:
it dont work for me please another solution...

marchande, 2022-01-30:
having the exact same problem here... just comenting so I can follow for the solution.

arwing123, 2022-01-30:
same problem

caall99, 2022-01-30:
Same issue here, but doing the SSH commands listed above with a couple restarts got me back to printing.

Arksine, 2022-01-30:
I believe I have a workaround.  SSH into your machine and run the following:
```
cd ~/moonraker
git pull
./scripts/install-moonraker.sh -r
```

If you haven't installed the policykit rules you may want to go ahead and do so now:
```
./scripts/set-policykit-rules.sh
```
Please respond with feedback if this worked for you.  Thanks, and I apologize for the downtime here.

Arksine, 2022-01-30:
Also, if you previously ran the install script with `-f` and are on FluiddPi/Mainsail OS you may need to slightly modify the options for the install script:
```
./scripts/install-moonraker.sh -r -f -c ~/klipper_config/moonraker.conf -l ~/klipper_logs/moonraker.log
```

Xaos1990, 2022-01-30:
> I believe I have a workaround. SSH into your machine and run the following:
> 
> ```
> cd ~/moonraker
> git pull
> ./scripts/install-moonraker.sh -r
> ```
> 
> **See the post below if you previously ran `install-moonraker.sh -f`**
> 
> If you haven't installed the policykit rules you may want to go ahead and do so now:
> 
> ```
> ./scripts/set-policykit-rules.sh
> ```
> 
> Please respond with feedback if this worked for you. Thanks, and I apologize for the downtime here.

This worked for me, thanks!

adrwhong, 2022-01-30:
> Also, if you previously ran the install script with `-f` and are on FluiddPi/Mainsail OS you may need to slightly modify the options for the install script:
> 
> ```
> ./scripts/install-moonraker.sh -r -f -c ~/klipper_config/moonraker.conf -l ~/klipper_logs/moonraker.log
> ```

Including the previous step, this part ultimately worked for me, thank you!

agent462, 2022-01-30:
> I believe I have a workaround. SSH into your machine and run the following:
>
> ```
> cd ~/moonraker
> git pull
> ./scripts/install-moonraker.sh -r
> ```
>
> **See the post below if you previously ran [...]

This also works for me.

FrYakaTKoP, 2022-01-30:
as already reported in discord seems to work fine 👍 

eric the hero of the "invalid doctype sunday" 

As i said before you need to make a kofi account or similar!

tommyl2210, 2022-01-30:
worked for me, too!! Thank you @Arksine for the fast support

ckvsoft, 2022-01-30:
> I believe I have a workaround. SSH into your machine and run the following:
>
> ```
> cd ~/moonraker
> git pull
> ./scripts/install-moonraker.sh -r
> ```
>
> **See the post below if you previously ran [...]

Thanks, It works ;-)

toby5419, 2022-01-30:
it works nice for the support----

stephbellemare, 2022-01-30:
> I believe I have a workaround. SSH into your machine and run the following:
>
> ```
> cd ~/moonraker
> git pull
> ./scripts/install-moonraker.sh -r
> ```
>
> **See the post below if you previously ran [...]

This worked for me as well, also, pip just got updated, which i upgraded as well. Unsure if it helped or not.
Thanks @Arksine

wissencat1, 2022-01-30:
> I believe I have a workaround. SSH into your machine and run the following:
>
> ```
> cd ~/moonraker
> git pull
> ./scripts/install-moonraker.sh -r
> ```
>
> **See the post below if you previously ran [...]

Saldy doesnt work for me

Arksine, 2022-01-30:
@stephbellemare Thanks.  FWIW, the 22.0.1 might cause issues the next time Moonraker needs to install a dependency.  I believe that 22.0.2 when released will resolve it, so keep an eye out for it and upgrade pip again when it is released.

Previously Moonraker always updated pip before fetching dependencies, however I have disabled that functionality for the time being.

Arksine, 2022-01-30:
@wissencat1 Was an error produced during the install process?  What is the output from:
```
sudo journalctl -xe -u moonraker
```

wissencat1, 2022-01-30:
I had to run it two times now it is working :)

stephbellemare, 2022-01-30:
> @stephbellemare Thanks. FWIW, pip 22.0.1 might cause issues the next time Moonraker needs to install a dependency. I believe that 22.0.2 when released will resolve it, so keep an eye out for it and upgrade pip again when it is released.
> 
> Previously Moonraker always updated pip before fetching dependencies, however I have disabled that functionality for the time being.

@Arksine I also just found out pip 22.0 also breaks Klipperscreen.
https://github.com/jordanruthe/KlipperScreen/issues/483

Quite a fun weekend

JWRu, 2022-01-30:
> cd ~/moonraker
> git reset --hard 9911b5c7dd2b8a8962a61e2cf9e98fa8fb1725c9
> sudo service moonraker restart

Worked for me.

semicuda, 2022-01-30:
> Please respond with feedback if this worked for you. Thanks, and I apologize for the downtime here.

Worked for me! Thank you :)

marchande, 2022-01-30:
sadly, none of this has worked for me... I have completely removed moonraker,  removed the directory,  then let kiauh re install moonraker  tried 
did a pull in the moonraker dir, and pulled the extra scripts that weren there before, installed  with -r and all reported back good... 
ran  ./scripts/set-policykit-rules.sh after re installing...  and still getting connection refused when trying to go to the page and klipper screen is still trying to connect the printer....

Arksine, 2022-01-30:
@marchande If Moonraker is running then its likely an unrelated issue.  To keep it separate from this thread, you may open up a new issue with a description, `klippy.log`, and `moonraker.log`.

FWIW, it looks like pip 22.0.1 does fix the issue.  I'll keep the hotfix in `install-moonraker.sh` until 22.0.2 is released and I can test it, at which point I think it will be safe to revert it.

marchande, 2022-01-30:
> 

> @marchande If Moonraker is running then its likely an unrelated issue. To keep it separate from this thread, you may open up a new issue with a description, `klippy.log`, and `moonraker.log`.
> 
> To all: It looks like pip 22.0.1 does fix the issue. I'll keep the hotfix in `install-moonraker.sh` until 22.0.2 is released and I can test it, at which point I think it will be safe to revert it.

@arksine, ultimatly after getting moonkraker working again,  I had to remove and re install mainsali and that fixed my problems,  wich is as you said a diferent problem but figured added an update here just incase any one else ends up with the same problem I did..

64bittuning, 2022-02-08:
Checking in. 
Just updated to 0.7.1-356 and got he ominous wall of text that Moonraker went stupid.

Ran the data above through SSH
`cd ~/moonraker
git pull
./scripts/install-moonraker.sh -r`

Issue was resolved. Rolled me back to -310.
FWIW, this install was done on mainsailOS and updates are done through the Update Manager in Mainsail. No additional scripts are run outside of that.

Arksine, 2022-02-08:
It shouldn't have been necessary to roll back.  The warningswere due an underlying change in Moonraker, see #349 for details..  The next time you update the warnings will return and you will have to take action to resolve them.  Alternatively you can stay at 0.7.1-310 if you would prefer.

64bittuning, 2022-02-09:
#349 Solved my issue, thanks.

Arksine, 2022-02-15:
I want to thank everyone for their feedback in this issue.  The pip issue has been resolved upstream, and going forward most users are going to face #349.  I'm going go ahead and close this issue to give bump 349 and give it more visibility.

closed 2022-02-15 by Arksine

JoeDiesAtTheEnd, 2022-12-13:
> I believe I have a workaround. SSH into your machine and run the following:
>
> ```
> cd ~/moonraker
> git pull
> ./scripts/install-moonraker.sh -r
> ```
>
> **See the post below if you previously ran [...]

**### this worked for me**

DarrenLindsey, 2022-12-27:
Very nice @JoeDiesAtTheEnd

switz-3d, 2023-01-22:
I'm the only one who this update has deinstalled pyton3?

TadasMit, 2023-02-08:
> 

Oh thank you! It was driving me crazy. This fixed my issue with moonraker too!

Xcion54, 2024-05-29:
> I believe I have a workaround. SSH into your machine and run the following:
>
> ```
> cd ~/moonraker
> git pull
> ./scripts/install-moonraker.sh -r
> ```
>
> **See the post below if you previously ran [...]

I had a mental breakdown trying to fix this, thank you so much