Arksine/moonraker #365: Updated moonraker

kshaub103, 2022-02-14:
I was updating and I when to updated moonraker and now and unable to print[image][image]

zellneralex, 2022-02-14:
See https://docs.mainsail.xyz/faq/klipper_errors/command_format_mismatch

Arksine, 2022-02-14:
Thanks Alex.  As mentioned, this is an issue with your Klipper installation.  You need to flash your mcu.

Arksine, 2022-02-15:
I'm going to close this since it isn't related to Moonraker.  If you need further assistance please visit the Klipper discord or discourse.

closed 2022-02-15 by Arksine