Arksine/moonraker #367: Moonraker do not update Raspberry Pi OS Packages after Recent Update

ymanyin, 2022-02-15:
### What happened

[image]

It was still working on v0.7.1-169-g50fd8c9.
However, update manager stopped working since recent 3-4 updates.
Thanks in advance for any help.

### Client

Fluidd

### Browser

Chrome, Firefox, Brave

### How to reproduce

Update to newest Moonraker.

masgar, 2022-02-15:
I had the same problem after I made a apt upgrade from command line. Fixed by calling the full update API.

From POSTMAN, or any other client able to make a POST, call http://your.rpi.name/machine/update/full

After a while you will get a result: "OK" response and all your packages will be OK.

Don't know if it's a bug but I fixed in this way.

Arksine, 2022-02-15:
Please see issue #349.  You need to grant Moonraker policykit permissions and make sure packagekit is installed.

closed 2022-02-15 by Arksine