Arksine/moonraker #373: on_when_upload_queued:True

jfle, 2022-02-19:
### What happened

I have setup a switch on Home Assistant. It works fine and i can turn on off my printer.  I had set the property on_when_upload_queued:True but when i upload from SuperSlicer (and i double chech that indeed the files is uploaded ) the printer does not turn on. 

### Client

Mainsail

### Browser

Chrome

### How to reproduce

Setup a power switch upload a print to the printer and wait to see if printer will boot.

Arksine, 2022-02-19:
Please attach moonraker.log reproducing the issue per the instructions in the issue template.  Also, make sure you followed all of the instructions [here](https://moonraker.readthedocs.io/en/latest/configuration/#power-on-g-code-uploads).

Arksine, 2022-03-16:
This issue has become inactive, so I am going to close it.  If there is still a problem you may reopen, please reproduce and attach moonarker.log.

closed 2022-03-16 by Arksine